apache/nuttx #73: The nxstyle is generating false alarm

acassis, 2020-01-10:
Hi Haitao Liu, when you have the chance, could you please verify the false alarm reported by nxstyle when handling empty block delimited by { } ?

Example:

  /* Bulk OUT endpoint descriptor */

  {
    int len = usbmsc_copy_epdesc(USBMSC_EPBULKOUT,
                                 (FAR struct usb_epdesc_s *)buf, devinfo,
                                 hispeed);

    buf += len;
    length += len;
  }

The alignment of the left and right braces are correct, but nxstyle get confused because there is not an if/while/switch, etc before the brace.

See more examples here:

=== drivers/usbdev/usbmsc_desc.c =============================================
drivers/usbdev/usbmsc_desc.c:342:2: error: Bad left brace alignment
drivers/usbdev/usbmsc_desc.c:343:4: error: Bad comment alignment
drivers/usbdev/usbmsc_desc.c:344:5: error: Bad comment block alignment
drivers/usbdev/usbmsc_desc.c:345:5: error: Bad comment block alignment
drivers/usbdev/usbmsc_desc.c:346:5: error: Bad comment block alignment
drivers/usbdev/usbmsc_desc.c:368:2: error: Bad right brace alignment
drivers/usbdev/usbmsc_desc.c:373:2: error: Bad left brace alignment
drivers/usbdev/usbmsc_desc.c:374:4: error: Bad alignment
drivers/usbdev/usbmsc_desc.c:390:2: error: Bad right brace alignment
drivers/usbdev/usbmsc_desc.c:396:2: error: Bad left brace alignment
drivers/usbdev/usbmsc_desc.c:402:2: error: Bad right brace alignment
drivers/usbdev/usbmsc_desc.c:406:2: error: Bad left brace alignment
drivers/usbdev/usbmsc_desc.c:413:2: error: Bad right brace alignment

Thank you very much in advance!

xiaoxiang781216, 2020-01-11:
@liuguo09 please take a look.

patacongo, 2020-01-11:
I don't think fixing these false alarms is time critical, but I think we should try to develop a full list of errors that nxstyle makes... both for fixing them in the future and so that people are aware of what kinds of false alarms they are expected to see.
I have a few more I will log too next time I see a good example.

patacongo, 2020-03-22:
I think we won't fix this.  I believe that people are now resolving this problem by increasing the indentation by two on this compound statements.  We may as well adopt that as the solution.   I will close this issue.

I do not see any place in the coding standard that addresses these "naked' compound statements.   There probably should be text added to the C coding statndard to indicate that the indentation for such compound statements follow the same indentation as, for example, compound statements that follow an "if" statement.

closed 2020-03-22 by patacongo