apache/nuttx #111: Can't work with CAN dev using internal OS interface

lukegluke, 2020-01-16:
In [drivers/can/can.c](https://github.com/apache/incubator-nuttx/blob/master/drivers/can/can.c) @walmis added support of multiple readers of the same port with 7d112f5639afef047ccf9aabf87e5c14d032a277, but in current state it breaks ability to work with CAN using internal OS interface.

Issue in current logic:
1) file_open() with reference to user-provided file structure *filep;
2) file_open() calls nx_vopen() that allocate a file structure for inode and call can_open() with reference to this file structure.
3) In can_open() https://github.com/apache/incubator-nuttx/blob/177c037aa9f0536c4d9baf25cf060c0633957b94/drivers/can/can.c#L475 can_reader is added to the list with the reference to file structure (from point 2). And returned
4) nx_vopen returns file descriptor
5) and that file descriptor is detached in file_open() if fact by duplicate the 'struct file' content into the user-provided file structure. 
6) then during reading in can_read() https://github.com/apache/incubator-nuttx/blob/177c037aa9f0536c4d9baf25cf060c0633957b94/drivers/can/can.c#L635 appropriate reader is chosen from list by filep value. But in this point filep argument of can_read() is the pointer to user-provided file structure, but the appropriate can_reader has pointer to another file structure (from point 2)! So can_read will not find any reader and fails this debug assertion if enabled: https://github.com/apache/incubator-nuttx/blob/177c037aa9f0536c4d9baf25cf060c0633957b94/drivers/can/can.c#L642

I'm not really good in all this things with inodes, files and descriptors, but as for me the solution will be to distinct readers in can driver not by filep itself, but by for example f_priv field of it.
Could we store value of filep in f_priv on can opening and use it in can_read? What happens with file structure (from point 2) with detaching, is there a chance this pointer be reused before file will be closed by user-provided reference?

p.s. maybe this issue also related to other devices, for instance serial

lukegluke, 2020-01-16:
I fix it as I suggested. It works now. But there is still important question about possibility that new reader can be added with file pointer that is already in readers list.

lukegluke, 2020-01-17:
Move to more simple solution: just storing pointer to specific reader in filep->f_priv

patacongo, 2020-01-17:
I believe that merging PR118 closes this issue.  Correct?

closed 2020-01-17 by patacongo

lukegluke, 2020-01-17:
Yes. Thanks!