apache/nuttx #127: missing DISCLAIMER file

justinmclean, 2020-01-19:
All ASF incubating projects need a DISCLAIMER file. One of two needs to be chosen see [1]

1. https://incubator.apache.org/policy/incubation.html#disclaimers

btashton, 2020-01-19:
Since I think we are striving to get a release out likely before we can say the licensing transition is fully complete we would need to go with the with "Work In Progress Disclaimer".  As part of that we need to list any known ways that the project is not compliant with Apache requirements.  @patacongo could you list any point here that you know of.  We can then move them into the file.

patacongo, 2020-01-19:
There is a lot of code that to speak authoritatively about without examining the files.

I am aware of possible problems with the licensing of the file in arch/arm/src/lpc43xx/spifi/src.  There are lots of issues with the code under that spifi directory.  It snuck in under the radar with a very large PR.  It also has coding standard problems.

There are only two C files there with some strange licensing.  I think it is BSD compatbile, but maybe not Apache 2 due to the NXP-only clause?

All other code is BSD-3 Compatible.  The COPYING file might not be fully up-to-date, but AFAIK still reflects the licensing used in the code.

justinmclean, 2020-01-19:
Hi,

> There is a lot of code that to speak authoritatively about without examining the files.

Great that make things a little easier.

> There are only two C files there with some strange licensing. I think it is BSD compatbile, but maybe not Apache 2 due to the NXP-only clause?

Any restriction of use to a hardware platform for instance would not be compatible  with the ALv2.

I just checked and yes "provided that it is used in conjunction with NXP Semiconductors microcontrollers.” is likely an issue. Does this specifically run on that platform or is it more generic code that could run on more than one platform?

MyMewt in a number of cases contacted the 3rd parties and got them to change their licensing or got permission to drop the restriction. You might also find that a more modern version of the file that the license has changed.

When using the WIP disclaimer you just note it as an issue to be resolved later.

Thanks,
Justin

btashton, 2020-01-19:
in the `arch/` folder there are 4669 total files, and 139 of them do not have licenses associated with them.  The vast majority are things like readme.txt, Kconfig, etc.. but it does include files like these ones that have a non-bsd license.  This should be fairly easy to address, but this is in quite good shape, which is a testament to how strict Greg has been on this point.

justinmclean, 2020-01-19:
Hi,

Yep I think are in good shape from a ALv2 compatibility point of view, just a couple of minor things to fix from what I’ve seen so far.

Apache has a simple tool for checking releases [1], you might want to run it as well. Although currently it just reports 12222 Unknown Licenses which is slightly less than helpful :-)

Do we want to get Fossology running on a VM at Infra? That we we can track changes between releases and the like.

Thanks,
Justin

1. https://creadur.apache.org/rat/

patacongo, 2020-01-19:
@justinmclean "I just checked and yes "provided that it is used in conjunction with NXP Semiconductors microcontrollers.” is likely an issue. Does this specifically run on that platform or is it more generic code that could run on more than one platform?"

I can only run on the specific LPC43 MCU.  There might be other non-tainted NXP SPIFI drivers for other NXP parts, that would require some significant effort to find out.

patacongo, 2020-01-19:
@btashton I did this for the IPMC in December, I think.  Maybe it is helpful.
[rep-stats.txt](https://github.com/apache/incubator-nuttx/files/4082704/rep-stats.txt)

closed 2020-03-05 by patacongo