apache/nuttx #268: to introduce a new function timerfd_settime() and timerfd_gettime()

edbek, 2020-02-13:
Is it possible to introduce a new function **timerfd_settime** () and **timerfd_gettime** ()?

patacongo, 2020-02-13:
I would prefer not.  These are non-standard GLIBC functions.  NuttX follows the standards defined at OpenGroup.org which does not include these interfaces.

There are exceptions to this in the code base now, but I don't think we should proliferate non-standard interfaces.

patacongo, 2020-02-13:
The solution would also required timerfd_create().  https://linux.die.net/man/2/timerfd_settime

While I don't advocate including non-standard interfaces in the OS, I would not oppose the inclusion of these GLIBC/Linux interfaces if they were properly implemented in the nuttx/fs directory (with new interfaces to sched/timer, were configurable, and had proper system calls implemented in syscall/.

edbek, 2020-02-13:
yes, of course, the timerfd_create () function is also required. I think it would be very useful for this project to have such a set of functions:

timerfd_create ();
thimerfd_settime ();
thimerfd_gettime ();

Thank you in advance if they appear !

xiaoxiang781216, 2020-02-13:
We can add eventfd and signalfd to the list too, these three API set is the good addition to POSIX since they strengthen the core UNIX design principle: Everything is a file.

edbek, 2020-02-13:
I agree !

patacongo, 2020-07-30:
eventfd APIs introduced with PR #1479.   PR #1479 does not address signalfd and timerfd interfaces, however.

eerimoq, 2020-09-10:
Hello!

I'm interested in `timerfd_create()` as well. Does anyone have an implementation ready? If not, I might try to implement it myself.

xiaoxiang781216, 2020-09-10:
@eerimoq as far as I know nobody start the work, it will be great to add timerfd support! Actually, you can reference the recent addition of eventfd PR #1479 and #1479.

pkarashchenko, 2021-12-17:
Trying to address this issue in scope of https://github.com/apache/incubator-nuttx/pull/5014

xiaoxiang781216, 2021-12-17:
Nice!

closed 2021-12-20 by xiaoxiang781216

xiaoxiang781216, 2022-12-28:
Ok, we support the final one(signalfd), great!