apache/nuttx #306: Cannot build NuttX with a rom image

sgysh, 2020-02-19:
I cannot build NuttX with a rom image.

```
CC:  stm32_romfs_initialize.c
{standard input}: Assembler messages:
{standard input}:498: Error: can't resolve `.rodata' {.rodata section} - `.Ltext0' {.text section}
{standard input}:499: Error: can't resolve `.rodata' {.rodata section} - `.Ltext0' {.text section}
{standard input}:502: Error: can't resolve `.rodata' {.rodata section} - `.Ltext0' {.text section}
{standard input}:503: Error: can't resolve `.rodata' {.rodata section} - `.Ltext0' {.text section}
/home/ysugino/incubator-nuttx/nuttx/boards/Board.mk:109: recipe for target 'stm32_romfs_initialize.o' failed
make[2]: *** [stm32_romfs_initialize.o] Error 1
make[2]: Leaving directory '/home/ysugino/incubator-nuttx/nuttx/boards/arm/stm32/stm32f4discovery/src'
Makefile:181: recipe for target 'board/libboard.a' failed
make[1]: *** [board/libboard.a] Error 2
make[1]: Leaving directory '/home/ysugino/incubator-nuttx/nuttx/arch/arm/src'
tools/Makefile.unix:465: recipe for target 'pass2' failed
make: *** [pass2] Error 2
```

## Steps to reproduce

1. Make a configuration

```
$ ./tools/configure.sh stm32f4discovery:rndis
```

2. Make a rom image

```
$ mkdir foo
$ cd foo/
$ echo bar > baz
$ genromfs -f rom.img
$ cp rom.img /PATH_TO_NUTTX_DIR/
```

3. Enable FS_ROMFS and STM32_ROMFS

```
$ make menuconfig
```

4. Build NuttX

```
$ make
```

## Workaround

I found a following workaround.

```
diff --git a/boards/arm/stm32/stm32f4discovery/src/stm32_romfs_initialize.c b/boards/arm/stm32/stm32f4discovery/src/stm32_romfs_initialize.c
index 9f63db65ad..79839b84c1 100644
--- a/boards/arm/stm32/stm32f4discovery/src/stm32_romfs_initialize.c
+++ b/boards/arm/stm32/stm32f4discovery/src/stm32_romfs_initialize.c
@@ -93,7 +93,8 @@ __asm__ (
 "romfs_data_end:\n"
     ".globl   romfs_data_size\n"
 "romfs_data_size:\n"
-    ".word romfs_data_end - romfs_data_begin\n");
+    ".word romfs_data_end - romfs_data_begin\n"
+    ".text\n");

 extern const char romfs_data_begin;
 extern const char romfs_data_end;
```

xiaoxiang781216, 2020-02-23:
@sgysh could you provide a PR for review?

sgysh, 2020-02-23:
I found the workaround, but I still don't understand the cause of this issue.

closed 2020-03-01 by xiaoxiang781216