apache/nuttx #438: drivers/video/isx012.c:  Make forbidden calls into CXD56xx code

patacongo, 2020-03-05:
Common drivers must never call directly in architecture specific code.  That is FORBIDDEN.

The driver at drivers/video/isx012.c does just that:

    $ grep cxd56 drivers/video/isx012.c
      ret = cxd56_cisifinit();
          imagererr("Fail cxd56_cisifinit %d\n", ret);
      ret = cxd56_cisifstopcapture();
          imagererr("Fail cxd56_cisifstopcapture %d\n", ret);
      ret = cxd56_cisiffinalize();
          imagererr("Fail cxd56_cisiffinalize %d\n", ret);
          ret = cxd56_cisifsetdmabuf(&sarea);
          ret = cxd56_cisifstartcapture(&cis_param, &sarea);
      ret =  cxd56_cisifstopcapture();

That is in complete violation of the requirements of the NuttX modular design.  This driver must either be corrected so that it uses a standard upper/lower half design as all other drivers do, or it must be removed, or it must be moved to  boards/arm/cxd56xx/drivers/

jerpelea, 2020-03-05:
I am looking into it ASAP

patacongo, 2020-03-05:
This was reported to me in a personal email from the same person that reported Issue #439.

closed 2020-03-07 by xiaoxiang781216