apache/nuttx #482: nxstyle:  header files that are not header files.

patacongo, 2020-03-07:
There is an issue with the graphics/nxglib/nxglib_fillrun.h header file (and probably others).

It is not really a header file. It is a "template" of a C file that is included with some definitions to make a complete C file. So it will NOT pass the tests for a header file.

See:

graphics/nxglib/lcd/nxglib_fillrectangle.c
graphics/nxglib/lcd/nxglib_filltrapezoid.c

It is not clear how this case should be handled. I believe we either have change the file name extension (but then it won't be checked at all) or change the extension to .c (which is misleading too).

Reference:  PR  #478

johannes-nivus, 2020-03-10:
Even if I state the obvious:
Name them .inc (or similar) and have nxstyle treat them as c file.
Should be a one liner...

patacongo, 2020-03-10:
> Even if I state the obvious:
> Name them .inc (or similar) and have nxstyle treat them as c file.
> Should be a one liner...

No, that was not obvious to me, but sounds like a good idea.  The extension .inc is atuallly used by some toolchains (the ZiLOG tools, it is an assembly language header file).