apache/nuttx #483: usbhost/usbhost_xboxcontroller.c compilation error

robin-raymond, 2020-03-07:
````
CC:  usbhost/usbhost_xboxcontroller.c
usbhost/usbhost_xboxcontroller.c: In function 'usbhost_notify':
usbhost/usbhost_xboxcontroller.c:561:36: error: 'struct usbhost_state_s' has no member named 'fds'
       FAR struct pollfd *fds = priv->fds[i];
                                    ^~
usbhost/usbhost_xboxcontroller.c: In function 'usbhost_poll':
usbhost/usbhost_xboxcontroller.c:2136:20: error: 'struct usbhost_state_s' has no member named 'fds'
           if (!priv->fds[i])
                    ^~
usbhost/usbhost_xboxcontroller.c:2140:19: error: 'struct usbhost_state_s' has no member named 'fds'
               priv->fds[i] = fds;
                   ^~
usbhost/usbhost_xboxcontroller.c:2141:35: error: 'struct usbhost_state_s' has no member named 'fds'
               fds->priv    = &priv->fds[i];
                                   ^~
usbhost/usbhost_xboxcontroller.c:2159:11: warning: implicit declaration of function 'usbhost_pollnotify' [-Wimplicit-function-declaration]
           usbhost_pollnotify(priv);
           ^~~~~~~~~~~~~~~~~~
````

Think it's missing:

````
  /* The following is a list if poll structures of threads waiting for
   * driver events. The 'struct pollfd' reference for each open is also
   * retained in the f_priv field of the 'struct file'.
   */

  struct pollfd *fds[CONFIG_HIDMOUSE_NPOLLWAITERS];
````

inside `struct usbhost_state_s`

But I'm not entirely sure if that's the right fix.

robin-raymond, 2020-03-07:
related -- I think a forward declaration is missing:
````
static void usbhost_pollnotify(FAR struct usbhost_state_s *dev);
````

acassis, 2020-03-07:
Hi Robin,
I think you fix is fine, could you please submit a PR ?

patacongo, 2020-03-07:
Resolved with PR #485

closed 2020-03-07 by patacongo