apache/nuttx #484: Remove CONFIG_FS_WRITABLE

patacongo, 2020-03-07:
This is kind of a useless configuration variable that occurs in many places and adds a lot of complexity and makes debug a lot more difficult.

It makes debug more difficult because it is not consistently set and requires a lot of debug time before you finally figure out "I needs to set CONFIG_FS_WRITABLE.

In most places affected by CONFIG_FS_WRITABLE, the configuration option is stupid:  Who need read-only MMC or read-only SmartFS.  Just not a useful think.  And many places that do need CONFIG_FS_WRITABLE do not automatically set it.  Like drivers/mtd/ftl.c.  In that case, you can debug for most of the day before you finally figure out why the you can't access flash via character driver (which is a really common configuration for resource limited platforms.

It is hard to image a platform that supports a file system but has no writing enabled.  I could image a system with only ROMFS or only CROMFS.  But that is really about it.  I am not aware of any such case, but that would be the only kind of platform that would suffer from the change.

But my opinion is that in the bigger picture, NuttX would be cleaner, less complex, and more usable with CONFIG_FS_WRITABLE removed.

xiaoxiang781216, 2020-03-08:
Agree, especially the saved code size without CONFIG_FS_WRITABLE is very smalled.

xiaoxiang781216, 2020-03-22:
Fix in apps: https://github.com/apache/incubator-nuttx-apps/pull/131

closed 2020-03-22 by xiaoxiang781216