apache/nuttx #510: Use lcd ssd1289 in stm32f4discovery failed

hanwenlei, 2020-03-10:
I use the config "CONFIG_EXAMPLES_LVGLDEMO=y, CONFIG_LCD_SSD1289=y and CONFIG_VIDEO_FB=y" in "stm32f4discovery:usbnsh", and can be compiled. But when I run the lvgldemo application in nsh, I get the error "ERROR: Failed to open /dev/fb0: 2". I don't know what went wrong. How can i solve this problem?

xiaoxiang781216, 2020-03-10:
> I use the config "CONFIG_EXAMPLES_LVGLDEMO=y, CONFIG_LCD_SSD1289=y and CONFIG_VIDEO_FB=y" in "stm32f4discovery:usbnsh", and can be compiled. But when I run the lvgldemo application in nsh, I get the [...]

2 means that lvgldemo can't find /dev/fb0: 
#define ENOENT              2
Please make sure you can see fb0 by:
ls /dev/
Otherwise, you have to fix fb0 issue before trying the lvgl demo.

hanwenlei, 2020-03-10:
When I do "ls /dev/", there really isn't "/dev/fb0". But in "boards\arm\stm32\stm32f4discovery\src\stm32_bringup.c", it calls "fb_register" function. I don't know why the fb driver is not registered.

xiaoxiang781216, 2020-03-10:
> When I do "ls /dev/", there really isn't "/dev/fb0". But in "boards\arm\stm32\stm32f4discovery\src\stm32_bringup.c", it calls "fb_register" function. I don't know why the fb driver is not registered.

You need add some log to debug, other people is hard to help here without the same enviornment.

patacongo, 2020-03-12:
This appears to be a user who needs help, not really an issue.  The more appropriate place to get help is in the dev@nuttx.apache.org mail list.  What do we need to do to close this?

xiaoxiang781216, 2020-03-13:
@hanwenlei I will close this issue. As @patacongo suggestion, you can post the new question to dev list, so more people can give you help.

closed 2020-03-13 by xiaoxiang781216