apache/nuttx #525: nxstyle:  Does not check for braces.

patacongo, 2020-03-10:
There is no check to assure that the statements following if, else, while, and for keywords are enclosed in braces.  This requirement is true if even if no statments follow or if only a single statement follows:  https://cwiki.apache.org/confluence/display/NUTTX/Coding+Standard#ifthenelse

Wrong:
      if (a == b)
        return;

      while (foobar == 0);

johannes-nivus, 2020-03-10:
I'd like to have a look concerning braces, but I'm not sure if I beat a dead horse. If someone is successful with a more common tool (as discussed on mailing list) all the work would become obsolete suddenly.

patacongo, 2020-03-10:
@johannes-nivus It is important that we document shortcomings in nxstyle, whether we fix them or not.  In order to implement this particular feature, some minimal syntax parsing would be required; I am not sure we want to go that route.  But we definitely need an issue in place to remind us that not all coding standard issues are addresses.

There are many, many things that are not checked by nxstyle.  Some that come to mind:

Non-local scope (global) varialbes must begin with g_
Structure names must end with _s, union names must end with _u, enumeration names must end with _e, typedef'e names must end with _t
All identifiers are local case except the pre-processor names are all upper case.
etc.

There are more coding standard issues that are NOT addressed than there are coding standard issues addresses.  I am not familiar with any automated pretty printer than can fix such things either.

xiaoxiang781216, 2020-03-10:
> @johannes-nivus It is important that we document shortcomings in nxstyle, whether we fix them or not. In order to implement this particular feature, some minimal syntax parsing would be required; I [...]

How about stdin/stdout/stderr? any naming rule will have exception, but the standard has the high priority than coding style.

> Structure names must end with _s, union names must end with _u, enumeration names must end with _e, typedef'e names must end with _t

But we need smart here: any struct/union/enum/typedef defined by the standard need to keep as before.

> All identifiers are local case except the pre-processor names are all upper case.
> etc.
> 
> There are more coding standard issues that are NOT addressed than there are coding standard issues addresses. I am not familiar with any automated pretty printer than can fix such things either.