apache/nuttx #536: unclear NuttXCCodingStandard examples

yamt, 2020-03-11:
NuttXCCodingStandard.html lists the following examples in both of "Preferred" and "Acceptable".

    extern int g_someint;
    static int g_anotherint;

davids5, 2020-03-11:
I think the problem is bigger then just this one.

Why do we have N ways?   Acceptable, Discouraged, Correct, Preferred,

In the past if you choose to do it the any way other then Preferred, your contribution would be modified to be the Preferred way.

This sort of thing is what makes having or creating a formatting tool a complex mess for NO ADDED value. 

Why not remove all the ambiguity from the fix the CS document and fix the tool?

patacongo, 2020-03-11:
We should carefully entertain any PR to modify the coding standard.  Remember that the version in Confluence is NOT the master.  The master is in the nuttx repository at Documents/NuttXCCCodingStandard.html and can be updated via PR as with any other file.  The verion on Confluence is then a copy of that.  Any changes made in Confluence will be lost the next time that the Documents are updated.

davids5, 2020-03-11:
@patacongo good to know - Should we add this note to the repo so that that never happens?

patacongo, 2020-03-11:
> Should we add this note to the repo so that that never happens?

It is true of all of the documents under the documents tabs.  These are only copies.  The really should just be read-only