apache/nuttx #587: pthread_mutex_lock() does not resume waiting after signal

hartmannathan, 2020-03-19:
According to the description of `pthread_mutex_lock()` (in the function's comment block):

"If a signal is delivered to a thread waiting for a mutex, upon return from the signal handler the thread resumes waiting for the mutex as if it was not interrupted."

But the code does not appear to do this.

`pthread_mutex_lock()` is a thin wrapper around `pthread_mutex_timedlock()` with a NULL abs_timeout. `pthread_mutex_timedlock()` calls `pthread_mutex_take()` with a intr value of true, which in turn calls `pthread_sem_take()` with intr value of true, which calls `nxsem_wait()`. `nxsem_wait()` returns -EINTR if awakened by a signal.

In the above mentioned callers, there does not appear to be logic to detect this case and resume waiting for the mutex as if it was not interrupted per the specification.

Notes:

(1) While the comment block above `pthread_mutex_timedlock()` does not make a similar statement as `pthread_mutex_lock()` regarding interruption by a signal, OpenGroup.org seems to [suggest](https://pubs.opengroup.org/onlinepubs/9699919799/functions/pthread_mutex_timedlock.html) that it should: "The pthread_mutex_timedlock() function shall lock the mutex object referenced by mutex. If the mutex is already locked, the calling thread shall block until the mutex becomes available as in the pthread_mutex_lock() function."

(2) See the dev@ mailing list thread entitled "pthread_mutex_lock() and signals" started 13 Mar 2020, archived here: [https://lists.apache.org/thread.html/rc366bf8ab1291e3f4d36ef90a2a2b4dc25d4f89db7ffc57df320e7ce%40%3Cdev.nuttx.apache.org%3E](https://lists.apache.org/thread.html/rc366bf8ab1291e3f4d36ef90a2a2b4dc25d4f89db7ffc57df320e7ce%40%3Cdev.nuttx.apache.org%3E)

patacongo, 2020-03-19:
NOTE:  Passing the value false to pthread_mutex_take() is not a solution.

That argument is passed to pthread_sem_take() and if the argument is false it calls:

    117       if (abs_timeout == NULL)
    118         {
    119           ret = nxsem_wait_uninterruptible(sem);
    120         }
    121       else
    122         {
    123           ret = nxsem_timedwait_uninterruptible(sem, abs_timeout);
    124         }

But nxsem_wait_uninterruptble does more than just ignore signal interupts:

    544 static inline int nxsem_wait_uninterruptible(FAR sem_t *sem)
    545 {
    546   int ret;
    547
    548   do
    549     {
    550       /* Take the semaphore (perhaps waiting) */
    551
    552       ret = nxsem_wait(sem);
    553     }
    554   while (ret == -EINTR || ret == -ECANCELED);
    555
    556   return ret;
    557 }

It also breaks thread cancellation.  When cancellation pointers are enabled, semaphore waits will be awakened with ECANCELED and must return.  In the deferred cancellation mode all resources will be cleaned-up and that task will terminated when it attempts to exit the cancellation point.

The above breaks thread cancellation because in the event that the thread is cancelled, it simply loops back and waits again.  The thread cannot be cancelled in the recommended deferred cancellation most.  That is seriaous OS breakage.

This is another bug in the existing implementation.  ALL semaphore waits must return in the event of thread cancellation, otherwise thread cancellation is broken (it is broken in many cases).

A better solution in  pthread_sem_take() would be to set up a sigprocmask and block relevant signals instead of calling nxsem_wait_uninterruptible().  That is too brutal.  All signals should be blocked except (1) the signal that wakes up the timed event and (2) any signals associated with thread cancellation.

patacongo, 2020-04-07:
> But nxsem_wait_uninterruptble does more than just ignore signal interrupts:

This has been fixed by a series of PRs culiminating in PR #733 

I think you could now proceed with this change without that particular concern raised in my comments.

closed 2024-10-07 by xiaoxiang781216