apache/nuttx #596: Add intree board config to test the relative CUSTOM board tooling

davids5, 2020-03-20:
See https://github.com/apache/incubator-nuttx/pull/594#issuecomment-601745637 for Context

In the past and recently changes in make files have broken the CUSTOM board tooling.
```
CONFIG_ARCH_BOARD_CUSTOM=y
CONFIG_ARCH_BOARD_CUSTOM_DIR="../nuttx-config"
CONFIG_ARCH_BOARD_CUSTOM_DIR_RELPATH=y
CONFIG_ARCH_BOARD_CUSTOM_NAME="tvc15-pro"
CONFIG_ARCH_CHIP="stm32h7"
CONFIG_ARCH_CHIP_STM32H743ZI=y
CONFIG_ARCH_CHIP_STM32H7=y
```

Adding a config will insure this does not continue to happen

patacongo, 2020-03-20:
> Adding a config will insure this does not continue to happen

And breakage of custom board configurations has occurred more than once in the past as well.

patacongo, 2020-04-12:
I've worked around this in a few files by putting multiple "Included Files" section headers.  That will eliminate the warning buf is probably not the right way to go.