apache/nuttx #606: nxstyle include check is too strict

patacongo, 2020-03-22:
As a rule, all #include pre-processor commands must appear in the "Included Files" section:  That section header must be in the file and all of the #include commands must follow.

However, there are exceptions.  For example:

    $ tools/nxstyle.exe arch/arm/include/xmc4/irq.h
    arch/arm/include/xmc4/irq.h:81:3: warning: #include outside of 'Included Files' section
    arch/arm/include/xmc4/irq.h:83:3: warning: #include outside of 'Included Files' section
    arch/arm/include/xmc4/irq.h:85:3: warning: #include outside of 'Included Files' section

This kind of problem is the case with many architectures. I have been ignoring this error in the past, but it is harder to ignore with this automated style checker.

And in this case, the #include cannot be moved because it depends on a #define that must precede it. And you cannot move the #define or your will get a similar error. I think that there is no alternative but you use good engineering judgement and accept the change as is.

This kind of logic is very common in architecure specific code:

    Set up some definitions, then
    include some architecture specific code

You will see this a lot in arch/xyz/include and arch/src/xyz/hardware directories.  This usually occurs int he "Pre-processor Definitions" section, and not in the "Included Files" section.

Ideally, then all #include's should be in the "Included Files" section. But there are exceptions where it makes sense to relax this requirement.  So example iss one and there are several others.

I don't know of any way to programmatically "relax" the requirement other than inspecting the nxstyle complaint and then making a good, informed engineering decision.  One possibility to to assure that "Included Files" was encountered sometime prior to the #include command.  That would allow the #include to occur in any section after the "Included Files" section but would have the odd side-effect of requiring that an "Included Files" section is present even when there are no other files to include.  That would, however, be rare.

yamt, 2020-03-23:
i had to make a change to similar construct: https://github.com/apache/incubator-nuttx/commit/63395b295f9820b4570e7327111f05a9683793e7

> And you cannot move the #define or your will get a similar error. 

really? in my case, i moved #define and it worked.

how about allowing an optional "Conditionally Included Files" section?

patacongo, 2020-04-12:
I've worked around this in a few files by putting multiple "Included Files" section headers. That will eliminate the warning buf is probably not the right way to go.