apache/nuttx #628: nxstyle: "Operator/assignment must be preceded with whitespace"

yamt, 2020-03-26:
nxstyle doesn't like the following line in apps/examples/sotest/sotest_main.c

      testfunc = (CODE void (*)(FAR const char *))dlsym(handle2, "testfunc1");

to me, the style seems perfectly fine.

xiaoxiang781216, 2020-03-26:
> nxstyle doesn't like the following line in apps/examples/sotest/sotest_main.c
> 
> ```
>   testfunc = (CODE void (*)(FAR const char *))dlsym(handle2, "testfunc1");
> ```
> 
> to me, the style seems perfectly fine.

I guess nxstyle like this:
```
 testfunc = (CODE void ( * )(FAR const char *))dlsym(handle2, "testfunc1");
```

yamt, 2020-03-26:
@xiaoxiang781216 
actually it doesn't.

    error: Space follows left parenthesis
    error: Space precedes right parenthesis

pkarashchenko, 2021-12-11:
Seems to be fixed by https://github.com/apache/incubator-nuttx/pull/4880

xiaoxiang781216, 2021-12-12:
So, let's close this issue.

closed 2021-12-12 by xiaoxiang781216