apache/nuttx #630: NuttX UINT32_C portability issue

PetervdPerk-NXP, 2020-03-26:
In stdint.h UINT32_C has not been defined.
Per section 7.20.4.1 of the ISO C11 standard, these macros shall be defined by a compliant implementation of the standard library. 
```
#if 0 /* REVISIT: Depends on architecture specific implementation */
#define INT8_C(x)           x
#define INT16_C(x)          x
#define INT32_C(x)          x ## l
#define INT64_C(x)          x ## ll

#define UINT8_C(x)          x
#define UINT16_C(x)         x
#define UINT32_C(x)         x ## ul
#define UINT64_C(x)         x ## ull
#endif
```

What would be a correct solution to define these macros for the corresponding CPU architecture?

PetervdPerk-NXP, 2020-06-30:
It seems that is issue has been fixed by this commit https://github.com/apache/incubator-nuttx/commit/7758eb865818d903dceeda4dd238985407673436, @xiaoxiang781216 that's correct right?

closed 2020-06-30 by PetervdPerk-NXP

xiaoxiang781216, 2020-06-30:
Yes, thanks for closing the issue.