apache/nuttx #652: NXStyle: Ignore alignments of block comments at right of code

Ouss4, 2020-03-30:
NXstyle rightfully ignores single comments at right of code, for instance within a struct declaration.
But complaints about a block comment.
```C
struct a_struct_s
{
   uint8_t f1; /* Long line in a single comment, OK. */
   uint8_t f2; /* Long line in a block comment,
                * nxstyle complains.
                */
};
```

patacongo, 2020-03-30:
The */ is normally not on a separate line on right hand comments.  Does this eliminte the problem?

    struct a_struct_s
    {
       uint8_t f1; /* Long line in a single comment, OK. */
       uint8_t f2; /* Long line in a block comment,
                    * nxstyle complains.  */
    };

Ouss4, 2020-03-30:
Tried this as well. I got the same complaints.

patacongo, 2020-04-25:
Can you please generate a test case?  I created this file, junk.c:

    /****************************************************************************
     * junk.c
     ****************************************************************************/
    
    /****************************************************************************
     * Included Files
    ****************************************************************************/
    
    #include <stdint.h>
    
    /****************************************************************************
     * Private Types
     ****************************************************************************/
    
    struct a_struct_s
    {
      uint8_t f1; /* Long line in a single comment, OK. */
      uint8_t f2; /* Long line in a block comment,
                   * nxstyle complains.
                   */
    };
    
    /****************************************************************************
     * Public Functions
     ****************************************************************************/
    
But nxstyle is completely happy with that... no complaints at all.  Nor have I ever seen a problem in the past.  So I am thinking that this is context specific problem.  So I think to resolve it, you will need to provide a full test file that shows the failure in context.

Is there an existing file in the repository?

patacongo, 2020-04-25:
Looking back through the PR chekcs for I think I see the issue you are referring to: https://github.com/apache/incubator-nuttx/runs/562410318?check_suite_focus=true

The file is  arch/arm/src/imxrt/imxrt_enc.c and these errors were reported:

    arch/arm/src/imxrt/imxrt_enc.c:266:78: error: Long line found
    arch/arm/src/imxrt/imxrt_enc.c:269:78: error: Long line found
    arch/arm/src/imxrt/imxrt_enc.c:272:82: error: Long line found
    arch/arm/src/imxrt/imxrt_enc.c:273:80: error: Long line found
    arch/arm/src/imxrt/imxrt_enc.c:296:78: error: Long line found

Looking at those, I see that these are not related to the right hand comments at all.  These are simply saying that the lines are too long.  I verfied each that this is the case.  The ARE too long.

When I run the current nxstyle against the current imxrt_enc.c I get:

    $ tools/nxstyle.exe arch/arm/src/imxrt/imxrt_enc.c
    arch/arm/src/imxrt/imxrt_enc.c:1037:81: error: Long line found
    arch/arm/src/imxrt/imxrt_enc.c:1043:84: error: Long line found

To things to note:

1. The errors at liness 266-296 no longer appear.  That is a regression.  They truly are too long.
2. New long lines are reported at 1037 and 1043.  These are correct and appear not due to a recent fix to nxstyle.

So I am thinking that this Issue should be closed.  Please give your feeback before I close it.

Ouss4, 2020-04-25:
Yes, the lines are too long.  But nxstyle ignores long lines when it's a comment at the right of struct field for instance.
This is used throughout the code base to comment struct fields, so the exception was added to nxstyle not too long ago.
The issue is, single line comments are ignored and no complaints is raised, however if it's a block (and long) comment, now nxstyle complains about the length.

patacongo, 2020-04-25:
> Yes, the lines are too long. But nxstyle ignores long lines when it's a comment at the right of struct field for instance.

I does not, but it did not use too.  That is a different error recently introduced into nxstyle when the right hand comment alignment logic was added.  We lost the length check then.

> This is used throughout the code base to comment struct fields, so the exception was added to nxstyle not too long ago.

It wasn't added intentionally.  It was an error that was introduced recently.  Now it turns out, I think it is a good behavior at least for now because there are two many long, right-hand comments, espectially int arch/ header definition header files.

> The issue is, single line comments are ignored and no complaints is raised, however if it's a block (and long) comment, now nxstyle complains about the length.

Let me modify my junk.c to make the lines too long and try again.

patacongo, 2020-04-25:
Nope, no long line is reported. Here is the modified junk.c file:

    /****************************************************************************
     * junk.c
     ****************************************************************************/
    
    /****************************************************************************
     * Included Files
     ****************************************************************************/
    
    #include <stdint.h>
    
    /****************************************************************************
     * Private Types
     ****************************************************************************/
    
    struct a_struct_s
    {
      uint8_t f1; /* This is a very long line in a single right-hand comment, No error is reported OK. */
      uint8_t f2; /* This is a a very long line in a block comment.  In this case nxstyle complains
                   * and an error is reported.
                   */
    };
    
    /****************************************************************************
     * Public Functions
     ****************************************************************************/
    
But nxstyle does not complain about the long lines at all.  I am not sure that that is good.  It is good for now, but there is no exception in the coding standard for long right hand comments.  They are also a violation of the coding standard, we just do not want to deal with them now:

The long right hand comments in **arch/arm/src/imxrt/imxrt_enc.c** no longer generate errors either.  So I conclude that this issue as reported no longer exists.  I believe that there are still issues wtih long lines:  They should be reported.  But given the context of this issue, it can be closed now.

closed 2020-04-25 by patacongo

Ouss4, 2020-04-25:
> Let me modify my junk.c to make the lines too long and try again.
I did so too.  And I confirm that nxstyle is working as it should be!

> Looking back through the PR chekcs for I think I see the issue you are referring to: https://github.com/apache/incubator-nuttx/runs/562410318?check_suite_focus=true
We can also see that some of the errors generated here no longer appear.  An example is line 272.