apache/nuttx #698: tools/testbuild.sh breaks non-git usage

hartmannathan, 2020-04-03:
Regarding commit [27142b2e3c4c6498dc198818ec9b260d8ffff12f](https://github.com/apache/incubator-nuttx/commit/27142b2e3c4c6498dc198818ec9b260d8ffff12f), this means
that tools/testbuild.sh now depends on git and on the NuttX
distribution being a git clone.

The problem with depending on git is that users who work with release
tarballs and/or don't have git can no longer use NuttX.

We should not clean with git unless doing so in a git clone. If not a git clone, we should invoke 'make distclean' as we were doing previously.

btashton, 2020-04-03:
Being that it is testbuild script I think that's less of an issue but this looks to me like it should fallback to make distclean if it cannot use git. Have you verified that that is not the case?

hartmannathan, 2020-04-03:
> Being that it is testbuild script I think that's less of an issue but this looks to me like it should fallback to make distclean if it cannot use git. Have you verified that that is not the case?

Yes. The logic to run 'make distclean' is removed.

hartmannathan, 2020-04-03:
Also per the mailing list discussion ([here](https://lists.apache.org/thread.html/r6ef57ae749f59d8013c789256df334a0f47945d1826ad95998ebfa43%40%3Cdev.nuttx.apache.org%3E)), this git clean invocation will remove other untracked files which the user may want to keep. I understand that it's a testbuild script but somewhere down the line someone will run it for whatever reason and it can cause a big problem.

btashton, 2020-04-03:
Maybe we can make a flag for that as an extra layer of protection, but the issue you opened is that this breaks non-git environments which does not look right to me.

btashton, 2020-04-03:
> > Being that it is testbuild script I think that's less of an issue but this looks to me like it should fallback to make distclean if it cannot use git. Have you verified that that is not the case?
> 
> Yes. The logic to run 'make distclean' is removed.

It was left. There is an OR issued now so if the git fails I would expect make distclean to run.

yamt, 2020-04-03:
* as @btashton btashton said, the fallback logic is still there. didn't it work for you?
* i submitted a patch to make the use of "git clean" optional. https://github.com/apache/incubator-nuttx/pull/701

hartmannathan, 2020-04-03:
> It was left. There is an OR issued now so if the git fails I would expect make distclean to run.

You're right, I stand corrected. Somehow I missed it before but now it looks obvious.

I think PR 701 is a good idea anyway, because of Greg's concern (in the mail thread I linked previously) that the git clean invocation would delete other non-versioned data which the user might want to keep.

btashton, 2020-04-03:
I agree, just wanted to make sure we had not missed something.

closed 2020-04-03 by xiaoxiang781216