apache/nuttx #844: Should the makefile warn if there isn't a valid APPDIR?

cmorganBE, 2020-04-22:
If you run the ./tools/configure.sh script without a ../apps folder or without setting APPDIR you get errors like below without any helpful reason what is going on. It took several minutes to figure it out.

```
nuttx$ ./tools/configure.sh olimex-stm32-e407:nsh
  Copy files
  Select CONFIG_HOST_LINUX=y
  Refreshing...
make: the '-C' option requires a non-empty string argument
Usage: make [options] [target] ...
Options:
...
```

Thoughts on having the Makefile check for this folder to provide reasonable warning/error messages?

patacongo, 2020-04-22:
It should be possible to build NuttX without any apps/ directory.  So this would be a bug.

cmorganBE, 2020-04-22:
@patacongo steps to reproduce here on Ubuntu 20.04 amd64:

```
mkdir nuttx_build_test
cd nuttx_build_test/
git clone git@github.com:apache/incubator-nuttx.git
cd incubator-nuttx/
./tools/configure.sh nucleo-h743zi:nsh
````

At this point I get:

````
 ./tools/configure.sh nucleo-h743zi:nsh
  Copy files
  Select CONFIG_HOST_LINUX=y
  Refreshing...
make: the '-C' option requires a non-empty string argument
Usage: make [options] [target] ...
Options:
  -b, -m                      Ignored for compatibility.
  -B, --always-make           Unconditionally make all targets.
  -C DIRECTORY, --directory=DIRECTORY
                              Change to DIRECTORY before doing anything.
  -d                          Print lots of debugging information.
  --debug[=FLAGS]             Print various types of debugging information.
  -e, --environment-overrides
                              Environment variables override makefiles.
  --eval=STRING               Evaluate STRING as a makefile statement.
  -f FILE, --file=FILE, --makefile=FILE
                              Read FILE as a makefile.
  -h, --help                  Print this message and exit.
  -i, --ignore-errors         Ignore errors from recipes.
  -I DIRECTORY, --include-dir=DIRECTORY
                              Search DIRECTORY for included makefiles.
  -j [N], --jobs[=N]          Allow N jobs at once; infinite jobs with no arg.
  -k, --keep-going            Keep going when some targets can't be made.
  -l [N], --load-average[=N], --max-load[=N]
                              Don't start multiple jobs unless load is below N.
  -L, --check-symlink-times   Use the latest mtime between symlinks and target.
  -n, --just-print, --dry-run, --recon
                              Don't actually run any recipe; just print them.
  -o FILE, --old-file=FILE, --assume-old=FILE
                              Consider FILE to be very old and don't remake it.
  -O[TYPE], --output-sync[=TYPE]
                              Synchronize output of parallel jobs by TYPE.
  -p, --print-data-base       Print make's internal database.
  -q, --question              Run no recipe; exit status says if up to date.
  -r, --no-builtin-rules      Disable the built-in implicit rules.
  -R, --no-builtin-variables  Disable the built-in variable settings.
  -s, --silent, --quiet       Don't echo recipes.
  -S, --no-keep-going, --stop
                              Turns off -k.
  -t, --touch                 Touch targets instead of remaking them.
  --trace                     Print tracing information.
  -v, --version               Print the version number of make and exit.
  -w, --print-directory       Print the current directory.
  --no-print-directory        Turn off -w, even if it was turned on implicitly.
  -W FILE, --what-if=FILE, --new-file=FILE, --assume-new=FILE
                              Consider FILE to be infinitely new.
  --warn-undefined-variables  Warn when an undefined variable is referenced.

This program built for x86_64-pc-linux-gnu
Report bugs to <bug-make@gnu.org>
make: *** [tools/Makefile.unix:364: dirlinks] Error 2
ERROR: failed to refresh
```

RockySong, 2020-11-20:
After I cloned https://github.com/apache/incubator-nuttx-apps and moved+renamed it to ../apps, I got other problems.
rs@DESKTOP-UG5S1LM:~/nuttx/incubator-nuttx$ ./tools/configure.sh ubw32:nsh
  Copy files
  Select CONFIG_HOST_LINUX=y
  Refreshing...
make[1]: Entering directory '/home/rs/nuttx/incubator-nuttx'
make[2]: Entering directory '/home/rs/nuttx/incubator-nuttx/boards'
make[2]: Leaving directory '/home/rs/nuttx/incubator-nuttx/boards'
make[2]: Entering directory '/home/rs/nuttx/apps'
make[3]: Entering directory '/home/rs/nuttx/apps/platform'
make[3]: Leaving directory '/home/rs/nuttx/apps/platform'
make[3]: Entering directory '/home/rs/nuttx/apps/builtin'
make[3]: Leaving directory '/home/rs/nuttx/apps/builtin'
make[2]: Leaving directory '/home/rs/nuttx/apps'
make[2]: Entering directory '/home/rs/nuttx/incubator-nuttx/graphics'
make[3]: Entering directory '/home/rs/nuttx/incubator-nuttx/graphics/nxglib'
make[3]: Leaving directory '/home/rs/nuttx/incubator-nuttx/graphics/nxglib'
make[3]: Entering directory '/home/rs/nuttx/incubator-nuttx/graphics/nxglib'
make[3]: Leaving directory '/home/rs/nuttx/incubator-nuttx/graphics/nxglib'
make[3]: Entering directory '/home/rs/nuttx/incubator-nuttx/graphics/nxglib'
make[3]: Leaving directory '/home/rs/nuttx/incubator-nuttx/graphics/nxglib'
make[2]: Leaving directory '/home/rs/nuttx/incubator-nuttx/graphics'
make[2]: Entering directory '/home/rs/nuttx/incubator-nuttx/arch/mips/src'
make[2]: Nothing to be done for 'clean_context'.
make[2]: Leaving directory '/home/rs/nuttx/incubator-nuttx/arch/mips/src'
make[1]: Leaving directory '/home/rs/nuttx/incubator-nuttx'
LN: include/arch to arch/mips/include
LN: include/arch/board to /home/rs/nuttx/incubator-nuttx/boards/mips/pic32mx/ubw32/include
LN: include/arch/chip to arch/mips/include/pic32mx
LN: arch/mips/src/board to /home/rs/nuttx/incubator-nuttx/boards/mips/pic32mx/ubw32/src
LN: arch/mips/src/chip to arch/mips/src/pic32mx
LN: /home/rs/nuttx/incubator-nuttx/drivers/platform to /home/rs/nuttx/incubator-nuttx/drivers/dummy
make[1]: Entering directory '/home/rs/nuttx/incubator-nuttx/libs/libxx'
make[1]: Nothing to be done for 'dirlinks'.
make[1]: Leaving directory '/home/rs/nuttx/incubator-nuttx/libs/libxx'
make[1]: Entering directory '/home/rs/nuttx/incubator-nuttx/boards'
make[1]: Leaving directory '/home/rs/nuttx/incubator-nuttx/boards'
make[1]: Entering directory '/home/rs/nuttx/incubator-nuttx/openamp'
make[1]: Nothing to be done for 'dirlinks'.
make[1]: Leaving directory '/home/rs/nuttx/incubator-nuttx/openamp'
make[1]: Entering directory '/home/rs/nuttx/apps'
make[2]: Entering directory '/home/rs/nuttx/apps/platform'
LN: platform/board to /home/rs/nuttx/apps/platform/dummy
make[2]: Leaving directory '/home/rs/nuttx/apps/platform'
make[1]: Leaving directory '/home/rs/nuttx/apps'
make[1]: Entering directory '/home/rs/nuttx/apps'
make[2]: Entering directory '/home/rs/nuttx/apps/examples'
/bin/sh: /home/rs/nuttx/apps/tools/mkkconfig.sh: Permission denied
make[2]: *** [/home/rs/nuttx/apps/Directory.mk:60: preconfig] Error 126
make[2]: Leaving directory '/home/rs/nuttx/apps/examples'
make[1]: *** [Makefile:167: Kconfig] Error 2
make[1]: Leaving directory '/home/rs/nuttx/apps'
make: *** [tools/Makefile.unix:604: apps_preconfig] Error 2
ERROR: failed to refresh

acassis, 2020-11-20:
Hi @RockySong the issue appears related to issues on your file system, could you please confirm that this file has execution permission, like mine:
```
$ ls -l apps/tools/mkkconfig.sh
-rwxrwxr-x 1 alan alan 2806 May  7  2020 apps/tools/mkkconfig.sh
```

RockySong, 2020-11-21:
Hi @acassis Noted with thanks for your reply!
Later I downloaded the 9.1.0 release from Nuttx official site and deleted the git cloned version, and 9.1.0 release worked normally. I checked permission and got below:

rs@DESKTOP-UG5S1LM:~/nuttx/nuttx_910$ ls -l apps/tools/mkkconfig.sh
**-rwxr-xr-x** 1 rs rs 2806 Jul  8 08:06 apps/tools/mkkconfig.sh

However, to verify the execution permission matters, then I cloned the git cloned back the app, and found it also has execution permission!

rs@DESKTOP-UG5S1LM:~/nuttx$ ls -l incubator-nuttx-apps/tools/mkkconfig.sh
**-rwxr-xr-x** 1 rs rs 2806 Nov 21 09:23 incubator-nuttx-apps/tools/mkkconfig.sh

It looks there are some other causes to this problem?

Robokishan, 2020-12-05:
I am having same problem on ubuntu and mac over nuttx git version i am just following documents as @RockySong mentioned may be there are some issues with git version

btashton, 2020-12-05:
@Robokishan I'm trying to understand what is going on here since we run all these builds against Ubuntu and Windows multiple times a day so I don't think git is broken.

There clearly is an issue if two people have run into to the same issue so I want to get to the root of it. 

If you clone directly to app `git clone https://github.com/apache/incubator-nuttx-apps apps` do you have an issue. This would eliminate something funny happening when you are moving things around (I don't see how that could cause an issue either).

The configure.sh script can also take a path to the apps folder so you do not have to rename it. 

Also what version of Ubuntu are you using?

Robokishan, 2020-12-05:
@btashton i am just following nuttx docs https://nuttx.apache.org/docs/latest/quickstart/quickstart.html

Robokishan, 2020-12-05:
After clonning it into app i am getting exactly the same errors what @RockySong had pasted

Robokishan, 2020-12-05:
and i checkout both nuttx-apps and nuttx to tag/nuttx-9.1.0 after that following errors

```
  Copy files
  Select CONFIG_HOST_MACOS=y
  Refreshing...
arch/arm/src/imxrt/Kconfig:1096: syntax error
arch/arm/src/imxrt/Kconfig:1095: invalid option
make: *** [olddefconfig] Error 1
ERROR: failed to refresh
```

Robokishan, 2020-12-05:
for a quick and dirty fix i checked out to nuttx 8.2 both from apps and main nuttx git

btashton, 2020-12-05:
@Robokishan see https://github.com/apache/incubator-nuttx/issues/2405  I am still not clear on what the issue was. But I think it was resolved using the kconfig that I pointed him at.

btashton, 2020-12-05:
I suspect you are somehow using an old version of kconfig when you run `which kconfig` do you get the version you built?

Robokishan, 2020-12-05:
Yes that's resolve the problem thanks for help!!!

btashton, 2020-12-05:
Great! You can install kconfig to a local path by passing `--prefix=` to `./configure.sh` which is what we do in the CI script here
./configure --prefix=$prebuilt/kconfig-frontends

Also just as a note 10.0.0 has been released (I need to send the announcement and update the site today).

btashton, 2020-12-05:
@v01d maybe we should add a note to the docs warning people about the brew version for macOS?

Robokishan, 2020-12-05:
Sure will checkout that also.  thanks for the help $prebuilt whould work ?

btashton, 2020-12-05:
> Sure will checkout that also. thanks for the help $prebuilt whould work ?

$prebuilt is a directory that we define at the top and then make sure $prebuilt/bin is in the PATH.  For my local development I tend to use `prefix=$HOME/.local/`  and then `export PATH=$PATH:$HOME/.local/bin` but everyone has there own way of managing local paths, so this is just an example.

v01d, 2020-12-05:
> @v01d maybe we should add a note to the docs warning people about the brew version of kconfig-fronteneds for macOS?

So the issue was there was already a system installed version? Maybe we need to add a message asking to uninstall that before a locally built one is used.

Robokishan, 2020-12-05:
@btashton not working it still trying to install it into the sytem. 
`./configure --prefix=$prebuilt/kconfig-frontends --enable-mconf --disable-shared --enable-static --disable-gconf --disable-qconf --disable-nconf`

without those options which are from docs.. it's giving me lots of errors so i added those options .

btashton, 2020-12-05:
Yeah sorry you do need to supply all the of those especially for local development.  Just make sure `$prebuilt/kconfig-frontends` point to where you want it installed for when you run `make install`

Robokishan, 2020-12-05:
same as yours i just want to install it to the $HOME/.local/bin so i exported it and then 
`export prefix=$HOME/.local/`
`./configure --prefix=$prebuilt/kconfig-frontends --enable-mconf --disable-shared --enable-static --disable-gconf --disable-qconf --disable-nconf`
`make install`
is there anything else i can add ?

btashton, 2020-12-05:
You should use --prefix=$HOME/.local/` instead of `--prefix=$prebuilt/kconfig-frontends`
then you will see the kconfig binaries in `$HOME/.local/bin/` after make install
`export PATH=$HOME/.local/bin:$PATH`  to make sure they are part of your path for the current bash session

y ~/.bashrc has this `export PATH=$HOME/.local/bin:$PATH` it is is always configured but how you manage this is up to you.  This is not NuttX specific.

Robokishan, 2020-12-05:
yes yes that's for sure.. now it is working thanks for the help that was the thing i was looking for i always use .local for the installation since sudo is too risky for anything.

Thanks for the help!

casaroli, 2026-11-08:
Possibly fixed by #5069.