apache/nuttx #845: Can we change armv8-m porting to apache license?

xiaoxiang781216, 2020-04-22:
@patacongo should we keep BSD license in the intial commit and wait you modify to apache license later?

patacongo, 2020-04-22:
I am not the expert here.  But I don't think that anything can come into the repositories that does not have an Apache 2.0 licence.  That is the rule that we have been enforcing.

If it is not Apache code, then it is third party code and I think we would need an SGA or some such thing.  Again, i am not the expert.  You should ask Justin, not me.

I would not personally merge any files that have BSD licenses or any Copyrights on them.

patacongo, 2020-04-22:
I think that all of the answers are here: https://www.apache.org/legal/resolved.html

xiaoxiang781216, 2020-04-22:
All most code under armv7-m come from you. It isn't a big issue If you give us the permission to modify the license.

patacongo, 2020-04-22:
Yes, you have my permission to modify the licenses to Apache 2.0 provided that I am the sole copyright holder and the sole author.  Other cases are more complex.

Perhaps we should resolve all license headers in armv7-m first?

patacongo, 2020-04-22:
Other than me and Pinecone as claimed copyright holders, these files look problematic:  dwt.h, etm.h, fpb.h, itm.h, itm_syslog.h, tpi.h, up_cache.c, up_itm.c, up_itm_syslog.c, up_stackcheck.c, and up_vectors.S

xiaoxiang781216, 2020-04-22:
> Yes, you have my permission to modify the licenses to Apache 2.0 provided that I am the sole copyright holder and the sole author. Other cases are more complex.
> 
> Perhaps we should resolve all license headers in armv7-m first?

Good ideal, we can resolve as much as possible first.

> Other than me and Pinecone as claimed copyright holders, these files look problematic: dwt.h, etm.h, fpb.h, itm.h, itm_syslog.h, tpi.h, up_cache.c, up_itm.c, up_itm_syslog.c, up_stackcheck.c, and up_vectors.S
> 

Pinecone is fully controlled by Xiaomi, so I think it isn't a big issue. How to handle the rest file, keep the copright as armv7-m?

> We need advice from a mentor.

btashton, 2020-04-22:
As has been mentioned in other threads, we cannot do this just based on the
copyright name. We also need to look at the authorship of the files.  This
needs to be done in a methodical way, there is some discussion on the wiki
about going forward with this.

Merging changes without doing this correctly makes it a lot harder to go
back and verify.

--Brennan

xiaoxiang781216, 2020-04-22:
Yes, that is why is better that @patacongo clean up armv7-m arch manually and then armv8-m just need follow the result. I think no other people better than him to decide the right copyright license.

patacongo, 2020-04-22:
> 
> 
> Yes, that is why is better that @patacongo clean up armv7-m arch manually and then armv8-m just need follow the result. I think no other people better than him to decide the right copyright license.

@xiaoxiang781216 Done.  Please review and merge PR #848

patacongo, 2020-04-22:
@xiaoxiang781216 There are a few Pinecone files that I could not change.  You wrote those files.  When you create new versions for armv8-m, you should be able to use the Apache 2.0 on the new files.

btashton, 2020-04-22:
Do we have a SGA for Pinecone. I think Justin has warned us about this before, that in this case we need the company approval to change the licensing of the file not just the author.

I'm all for getting this done I just don't want to have questions later about if the conversion was done correctly.

patacongo, 2020-04-22:
> Do we have a SGA for Pinecone. I think Justin has warned us about this before, that in this case we need the company approval to change the licensing of the file not just the author.

The is a different issue.  We did not change any Pinecone headers and do not plan to.

That is for the ARMv7-M architecture.  Xiang is implementing the ARMv8-M architecture which will have all new files.  If Xiang wrote those files and put the Pinecone license on them.  If Xiang generates new files they will have Apache licensing.

So this is a grey area.  But rest assured, not existing Pinecone file will have its header changed.

patacongo, 2020-04-22:
I am the only party who has submitted the SGA so only my files can have the header changed.

btashton, 2020-04-22:
Awesome! This all sounds good to me, just wanted to make sure we are on the same page to save trouble down the road.

patacongo, 2020-04-22:
There are a handful of other problem files.  I left them as they are for ARMv7-M.  The question is, "What do we do for those when we leverage them to ARMv8-M?"  There are three ARM header files, one PX4 file, and three or four more with either other people's persoanan name in the Copyright.  Or then made trivial changes to my file and then decided they deserve a Copyright too.  I always thought that was bad taste but that is the way it is.

xiaoxiang781216, 2020-04-23:
Ok, we will refine our patch basse on @patacongo copyright change, thanks.

closed 2020-04-23 by xiaoxiang781216