apache/nuttx #854: A error in sched\timer\timer_gettime.c?

yanyunyouyou, 2020-04-23:
89:  clock_ticks2time(timer->pt_last, &value->it_interval);
Why use timer->pt_last to set value->it_interval,value->it_interval is the period value.
so maybe use timer->pt_delay  is right!
[test.txt](https://github.com/apache/incubator-nuttx/files/4523259/test.txt)

xiaoxiang781216, 2020-04-24:
@yanyunyouyou yes, you are right, could you provide a patch?

yanyunyouyou, 2020-04-24:
[0001-timer_gettimee.c-89-change-the-timer-pt_last-to-time.patch.txt](https://github.com/apache/incubator-nuttx/files/4526947/0001-timer_gettimee.c-89-change-the-timer-pt_last-to-time.patch.txt)

xiaoxiang781216, 2020-04-24:
@yanyunyouyou could you follow the steps described here to generate a new PR:
https://help.github.com/en/github/collaborating-with-issues-and-pull-requests/github-flow
then we can work more efficiently.

closed 2020-04-24 by xiaoxiang781216