apache/nuttx #858: Would a PR to switch to '#pragma once' be accepted?

cmorganBE, 2020-04-24:
Would switch the #ifdef, #define, #endif entries to a single #pragma once at the top of each header file. #pragma once is widely supported at this point.

xiaoxiang781216, 2020-04-24:
I think it's hard to use this since:
1.NuttX need comatablie with C89 standard
2.#pragma once isn't defined in C standard
I also don't see the benefit to switch #ifndef/#define/#endif to #pragma once, do you have any particular reason?

cmorganBE, 2020-04-24:
@xiaoxiang781216 just seeing a lot of complex header protection defines that someone has to put together (or script) that would go away.

@patacongo I appreciate that, normally I wouldn't advocate for a pragma as they are compiler implementation specific. It's too bad that it hasn't made its way into the language standard proper yet. It's been supported by basically all compilers for a dozen years now.

If, at some point, an exception would be made for this particular pragma, let me know!

closed 2020-04-24 by cmorganBE

patacongo, 2020-04-24:
No worth it.  There is no real benefit and a very, very big downside.  We must not introduce ANY compiler dependencies in the source code.  We have followed that rule for a dozen years and it will not change now..  Let's not consider this.

patacongo, 2020-04-24:
Very bad idea.