apache/nuttx #886: Cygwin sim:nsh failed on apps/builtin

btashton, 2020-04-27:
Did a fresh install of cygwin and checked out the releases/9.0 branch of nuttx and apps. 
```
$ ./tools/configure.sh -c sim:nsh
$ make
<CUT>
make[1]: Entering directory '/home/btashton/apache/apps'
make[2]: Entering directory '/home/btashton/apache/apps/builtin'
./exec_builtin.c:54:10: fatal error: builtin/builtin.h: No such file or directory
   54 | #include "builtin/builtin.h"
      |          ^~~~~~~~~~~~~~~~~~~
compilation terminated.
ERROR: cc failed: 1
       command: cc -MT ./exec_builtin.home.btashton.apache.apps.builtin.o  -M -Wall -Wstrict-prototypes -Wshadow -Wundef -g -fno-builtin -fno-common -I. -isystem /home/btashton/apache/nuttx/include -D__KERNEL__ -pipe -I C:\cygwin64\home\btashton\apache\apps\include ./exec_builtin.c
make[2]: *** [/home/btashton/apache/apps/Application.mk:224: .depend] Error 1
make[2]: Leaving directory '/home/btashton/apache/apps/builtin'
make[1]: *** [Makefile:67: /home/btashton/apache/apps/builtin_depend] Error 2
```

`nuttx.exe` is still created and runs.  I did not see this targeting Linux using WSL.

patacongo, 2020-04-27:
The Windows builds are getting more quirky all of the time.  I opened a couple of other issues for weird behavior too.  Basic problem is that the people who are modifying the build system use Linux.  Build testing includes only Linux, but has recently been updated to cover macOS, but no Windows platforms.

This is very big exposure because over 60% of the people who download NuttX (and presumably use it) are on Windows platforms.  More than both Linux and macOS combined.  Yet there is no development or testing on the most common NuttX user platform.

Reference: https://sourceforge.net/projects/nuttx/files/stats/os?dates=2007-02-01%20to%202020-04-01&period=monthly

patacongo, 2020-04-27:
> `nuttx.exe` is still created and runs. I did not see this targeting Linux using WSL.

BTW:  That is because the failure occurred during dependency generation.  It is not a normal compilation failure so the build continued normally, but without proper dependencies.

The inclusion is for the file apps/include/builtin/builtin.h which exists.  So the problem in is the include path:  -I. -isystem /home/btashton/apache/nuttx/include -I C:\cygwin64\home\btashton\apache\apps\include

Oddly, of the two abslute paths, one is a cygwin POSIX path and the other is Windows native path.

I don't what toolchain you are using, but one of those is wrong.  A Cygwin tool would be happy with the first; a Windows native tool would be happy with the second.  But it is the second which is not working.

This is an error in the apps/ build system.

patacongo, 2020-04-27:
> A Cygwin tool would be happy with the first; a Windows native tool would be happy with the second. But it is the second which is not working.

The decision to convert paths from Cygwin POSIX paths to Windows native paths is determined by the setting WINTOOL.

> I don't what toolchain you are using, but one of those is wrong.

Since you are building on the simulator, you should be using the Cygwin gcc x86_64 compiler which would always want POSIX paths and WINTOOL should not be defined.

patacongo, 2020-04-27:
Just checking because I don't recall seeing that error in the past.  Perhaps it was introduced recently.  Anyway, here is a replication with a little more verbose make output:

    $ tools/configure.sh -c sim:nsh
    $ make V=1
    
    make[2]: Entering directory '/cygdrive/d/Spuda/Documents/projects/nuttx/master/incubator_apps/builtin'
    rm -f  .xx_builtin_list.h
    touch .xx_builtin_list.h
    cat  registry/hello.bdat registry/nsh.bdat registry/sh.bdat >  .xx_builtin_list.h
    mv .xx_builtin_list.h builtin_list.h
    rm -f  .xx_builtin_proto.h
    touch .xx_builtin_proto.h
    cat  registry/hello.pdat registry/nsh.pdat registry/sh.pdat >  .xx_builtin_proto.h
    mv .xx_builtin_proto.h builtin_proto.h
    /cygdrive/d/Spuda/Documents/projects/nuttx/master/incubator_nuttx/tools/mkdeps --dep-path . --obj-path . --obj-suffix .cygdrive.d.Spuda.Documents.projects.nuttx.master.incubator_apps.builtin.o "cc" -- -Wall -Wstrict-prototypes -Wshadow -Wundef -g -fno-builtin -fno-common -I. -isystem /cygdrive/d/Spuda/Documents/projects/nuttx/master/incubator_nuttx/include  -D__KERNEL__  -pipe -I "D:\Spuda\Documents\projects\nuttx\master\incubator_apps\include" -- builtin_list.c exec_builtin.c >Make.dep
    ./exec_builtin.c:54:10: fatal error: builtin/builtin.h: No such file or directory
       54 | #include "builtin/builtin.h"
          |          ^~~~~~~~~~~~~~~~~~~
    compilation terminated.
    ERROR: cc failed: 1
           command: cc -MT ./exec_builtin.cygdrive.d.Spuda.Documents.projects.nuttx.master.incubator_apps.builtin.o  -M -Wall -Wstrict-prototypes -Wshadow -Wundef -g -fno-builtin -fno-common -I. -isystem /cygdrive/d/Spuda/Documents/projects/nuttx/master/incubator_nuttx/include -D__KERNEL__ -pipe -I D:\Spuda\Documents\projects\nuttx\master\incubator_apps\include ./exec_builtin.c
    make[2]: *** [/cygdrive/d/Spuda/Documents/projects/nuttx/master/incubator_apps/Application.mk:224: .depend] Error 1
    make[2]: Leaving directory '/cygdrive/d/Spuda/Documents/projects/nuttx/master/incubator_apps/builtin'

btashton, 2020-04-27:
> The Windows builds are getting more quirky all of the time. I opened a couple of other issues for weird behavior too. Basic problem is that the people who are modifying the build system use Linux. [...]

Yeah I am going to submit a PR to add CI for windows now that I have an environment to test in.  It should actually not be to bad, but will have to wait until after the RC is out for voting.  The native build will take a bit more work to get wired up correctly, but the Cygwin one was really not bad at all.

For reference this is how I setup the environment:
Minimal Cygwin with these packages added:
```
    make              bison             libmpc-devel
    gcc-core          byacc             automake-1.15
    gcc-g++           gperf             libncurses-devel
    flex              gdb               libmpfr-devel
    git               unzip             zlib-devel
```
Then I built the kconfig-frontend like we do for all the other platforms.

I also did a test with an ARM configuration and for that I just grabbed the zip install from here:
https://developer.arm.com/-/media/Files/downloads/gnu-rm/9-2019q4/gcc-arm-none-eabi-9-2019-q4-major-win32.zip

xiaoxiang781216, 2020-04-27:
The similar error also happen with Ubuntu but very rarely. I suspect that the builtin.h generation hit some race condition.

patacongo, 2020-04-27:
> Yeah I am going to submit a PR to add CI for windows now that I have an environment to test in. It should actually not be to bad, but will have to wait until after the RC is out for voting. The [...]

Cygwin is time-consuming to install and is not as fast as Linux, but it has proven to be a solid, reliable platform.  I have not used Linux in years (except for as a test platform).  Most of NuttX was developed on Cygwin and just works on Linux.

> For reference this is how I setup the environment:
> Minimal Cygwin with these packages added:
>
> ```
>     make              bison             libmpc-devel
>     gcc-core          byacc [...]

This is good information.  I might be helpful in the top-level readme.

There are other things you would probably need later, like patch.  I have a fairly minimal Cygwin install on my laptop.  It is the default development system plus the packages that I needed to build and work with the code.  I am attaching a list of all installed packages (most were auto-installed  with the basic development system):  
[packages.txt](https://github.com/apache/incubator-nuttx/files/4537057/packages.txt)

> Then I built the kconfig-frontend like we do for all the other platforms.

And you might need a few other things like genromfs.  There is a snapshot here: https://bitbucket.org/nuttx/tools/src/master/

> I also did a test with an ARM configuration and for that I just grabbed the zip install from here:
> https://developer.arm.com/-/media/Files/downloads/gnu-rm/9-2019q4/gcc-arm-none-eabi-9-2019-q4-major-win32.zip

That is a Windows native toolchain.  It does need the black magic to convert paths to Windows paths.  WINTOOL should be selected by arch/arm/src/*/Toolchain.defs

You can build a Cygwin native ARM cross-compiler here:  https://bitbucket.org/nuttx/buildroot/src/master/  It deals naturally with the Cygwin POSIX paths.

patacongo, 2020-04-27:
> 
> 
@xiaoxiang781216 

> The similar error also happen with Ubuntu but very rarely. I suspect that the builtin.h generation hit some race condition.

This is not the autogenerated builtin*.h files, this is the repositories, fixed header file at apps/include/builtin/builtin.h.  It is always present.  The failure is occuring because the CFLAGS are wrong.  From above

/cygdrive/d/Spuda/Documents/projects/nuttx/master/incubator_nuttx/tools/mkdeps --dep-path . --obj-path . --obj-suffix .cygdrive.d.Spuda.Documents.projects.nuttx.master.incubator_apps.builtin.o "cc" -- -Wall -Wstrict-prototypes -Wshadow -Wundef -g -fno-builtin -fno-common -I. -isystem /cygdrive/d/Spuda/Documents/projects/nuttx/master/incubator_nuttx/include  -D__KERNEL__  -pipe **-I "D:\Spuda\Documents\projects\nuttx\master\incubator_apps\include"** -- builtin_list.c exec_builtin.c >Make.dep

The failure is occuring because a Windows native path, presumably created via cygpath, is being used again the Cygwin gcc which requires POSIX style paths.  As a result, the include path is ignored and it fails to find builtin.h at apps/include/builtin.  This is a problem in either Application.mk or in nuttx/tools/mkdep.c

patacongo, 2020-04-27:
I found the problem.  It was an error introduced by Xiang in tools/incdir.sh by commit 3b9efc95a2e on January 6, 2020.  PR is with more information is forthcoming.

closed 2020-04-27 by Ouss4

patacongo, 2020-04-27:
Issue #386 is still a problem.  I think this occurs when you configure on a Windows Cygwin or MSYS but the defconfig is Linux.  It will try to convert the configure to Windows Cygwin or MSY but the make oldconfig then generates warnings.

This was not a problem in the past because kconfig-tweak was used to modify defconfig files.  That was replaced with sed.   kconfig-tweak is safe and always generates a correct configuration.  sed is not safe and creates invalid defconfig configurations that generate errors when we try to produce the .config (via make olddefconfig)

patacongo, 2020-04-27:
@btashton @xiaoxiang781216 I am trying to test pr #897 but there is a problem with the macOS builds.  When the kconfig-frontend package was build, it did not enable kconfig-tweak which is one of the kconfig-frontend utilities that must be built for the build to be successful.

We cannot use sed.  We must use kconfig-tweak!

btashton, 2020-04-27:
@patacongo  I think if you add it to the cibuild script in testing it will rebuild the tool with this and invalidate the cache. 

I just about have a CI run working for Cygwin but it is ugly and really slow. I'll put up a Draft PR tonight and we can work on getting the Windows testing in a much better place. 

It has been a while since I have messed with Windows your help is appreciated.

patacongo, 2020-04-27:
The problem is here:

running CONFIG_SHELL=/bin/sh /bin/sh ./configure --prefix=/Users/runner/runners/2.169.1/work/incubator-nuttx/incubator-nuttx/sources/testing/../prebuilt/kconfig-frontends --disable-kconfig --disable-nconf --disable-qconf --disable-gconf --disable-mconf --disable-static --disable-shared --disable-L10n --disable-utils --no-create --no-recursion

--disable-utils suppresses building of all kconfig-frontend utils.  That is a very bad idea.  For one thing, it suppresses kconfig-tweak and cause this problem.

patacongo, 2020-04-27:
@btashton Thanks for the tip.  Can you please review and merge this:  https://github.com/apache/incubator-nuttx-testing/pull/38

Thanks

patacongo, 2020-04-27:
> I just about have a CI run working for Cygwin but it is ugly and really slow. I'll put up a Draft PR tonight and we can work on getting the Windows testing in a much better place.

I think we should be the PR checks short and sweet.  Just enough to give us some confidence that the change is not generally lethal.  To me this could be primarily Linux tests of various configurations.  More exhaustive testing should be left for the nightly build.

If a Windows CI is really time consuming too much time for efficent PR checks, perhaps only a nightly test would be sufficient.  That would not detect breakage on the PR, but would still catch the breakage post-merge.  Which, at least to me, is and acceptable performance/thoroughness tradeoff.

btashton, 2020-04-27:
> @btashton Thanks for the tip.  Can you please review and merge this:  https://github.com/apache/incubator-nuttx-testing/pull/38
> 
> Then I will restart the checks of #897 and maybe we can have a clean Windows Cygwin build.
> 
> Thanks

Checked it rebuilt the tools and merged it.

xiaoxiang781216, 2020-04-28:
> I think we should be the PR checks short and sweet. Just enough to give us some confidence that the change is not generally lethal. To me this could be primarily Linux tests of various [...]

Yes, this is how macOS do by @yamt, which just build three config(arm12, sim and mips-riscv-x86-xtensa).
For Windows, I would suggest that:
1.Build sim for all variants(Cygwin, Msys2, Native and WSL) since sim tightly couple with OS.
2.Select one but different arch for each variant
3.Enable all config which can only build on Windows(e.g. z80?)

> If a Windows CI is really time consuming too much time for efficent PR checks, perhaps only a 

Now, we build on Linux/macOS with 18VMs, but the finish time is around from 7m to 41m. So we have enough room to optimize VMs usage(distrube the work more balance) to 16VMs at least, then we have 4VMs for each Windows variants.

> nightly test would be sufficient. That would not detect breakage on the PR, but would still catch the breakage post-merge. Which, at least to me, is and acceptable performance/thoroughness tradeoff.

It's hard to deploy the nightly build to Apache Windows server, since:
1.Apache Windows server is much less than Linux server
2.We don't have the right to install software directly without JNFRA help

btashton, 2020-04-28:
> Yes, this is how macOS do by @yamt, which just build three config(arm12, sim and mips-riscv-x86-xtensa).
> For Windows, I would suggest that:
> 1.Build sim for all variants(Cygwin, Msys2, Native and [...]

Just for reference it currently takes over 1.5hr to run the Cygwin sim builds.  There are a couple optimizations I can make but I would not expect to shave more than 30%.  WSL is just Ubuntu so I don't see any reason to duplicate that. 

The Native builds should be quite a bit faster.

> 2.Select one but different arch for each variant
> 3.Enable all config which can only build on Windows(e.g. z80?)
>
>
> Now, we build on Linux/macOS with 18VMs, but the finish time is around from 7m to [...]

I don't really see any reason we cannot do the nightly build using GitHub Actions we can easily make the targets conditional on the job and then we don't have to redo all the CI work and have close to full coverage.

xiaoxiang781216, 2020-04-28:
> Just for reference it currently takes over 1.5hr to run the Cygwin sim builds. There are a couple optimizations I can make but I would not expect to shave more than 30%. 

1.5hr*70%=1hr, it's still a little bit slow, it will be great if we can finish in 40m

> WSL is just Ubuntu so I don't see any reason to duplicate that.

Yes, we can skip WSL. I use both Ubuntu and WSL daily, no problem specific for WSL find so far.
 
> The Native builds should be quite a bit faster.
>
>
> I don't really see any reason we cannot do the nightly build using GitHub Actions we can easily make the targets conditional on the job and then we [...]

Good point, we can run the nightly build with GitHub too. But since Apache server run on the real hardware with extreme large number cpu core, it is very useful to find the parallel build issue.