apache/nuttx #951: libc ioctl not compatible with 64bit hardware

btashton, 2020-05-03:
The iotctl function requires that `void *` `int` both be equal to `unsigned long` but in the case of 64 bits hardware like x86_64 this may not be the case. It seems like we should be able to make this work by just checking to see if `int` will fit in `unsigned long` rather than checking that it is exactly unsigned long.

```
int ioctl(int fd, int req, ...)
{
  unsigned long arg;
  va_list ap;

  /* Get the unsigned long argument.
   *
   * REVISIT:  This could be the cause of the crash down the road if the
   * actual size of the argument is anything other than sizeof(unsigned long).
   * Most small integers will be promoted to 'int'.  ARM should pass the
   * following test with all three types having sizeof(type) == 4 bytes.
   * 'float' should also be tested.  But 'long long' and 'double' are out of
   * the question!  Don't try to pass them.
   *
   * And what will happen if no third argument is passed?  In most cases,
   * this should just result in a garbage value for arg.  But you may
   * discover cases where something worse happens!
   */

  DEBUGASSERT(sizeof(int)        == sizeof(unsigned long) &&
              sizeof(FAR void *) == sizeof(unsigned long));

  va_start(ap, req);
  arg = va_arg(ap, unsigned long);
  va_end(ap);

  /* Then let fs_ioctl() to the real work */

  return fs_ioctl(fd, req, arg);
}
```

btashton, 2020-05-03:
@patacongo Can we change this debug assert to `sizeof(int)        <= sizeof(unsigned long) ` without causing issues?

patacongo, 2020-05-03:
> Can we change this debug assert to `sizeof(int) <= sizeof(unsigned long) ` without causing issues?

This could actually cause issues, couldn't it?  But there are a lot of issues now that we have made the variadic ioctl() the only interface.  I will submit an Issue.

patacongo, 2020-05-03:
i just create Issue #959 to address the problems that I see and the only solution that I see.

patacongo, 2020-05-03:
Hmmm... this is an Issue, not a PR.  I guess I didn't need to create another issue.

btashton, 2020-05-03:
I'll mark this as a duplicate so we can track it in your more detailed ticket.
Duplicate of #959

btashton, 2020-05-03:
Duplicate of #959

closed 2020-05-03 by btashton