apache/nuttx #954: x86_64 fails ostest if PRIORITY_INHERITANCE is enabled

btashton, 2020-05-03:
x86_64 fails ostest if PRIORITY_INHERITANCE is enabled
```
user_main: semaphore test
sem_test: Initializing semaphore to 0
sem_test: Starting waiter thread 1
sem_test: Set thread 1 priority to 191
waiter_func: Thread 1 Started
waiter_func: Thread 1 initial semaphore value = 0
waiter_func: Thread 1 waiting on semaphore
sem_test: Starting waiter thread 2
sem_test: Set thread 2 priority to 128
waiter_func: Thread 2 Started
waiter_func: Thread 2 initial semaphore value = -1
waiter_func: Thread 2 waiting on semaphore
sem_test: Starting poster thread 3
sem_test: Set thread 3 priority to 64
poster_func: Thread 3 started
poster_func: Thread 3 semaphore value = -2
poster_func: Thread 3 posting semaphore
waiter_func: Thread 1 awakened
waiter_func: Thread 1 new semaphore value = -1
waiter_func: Thread 1 done
poster_func: Thread 3 new semaphore value = -1
poster_func: Thread 3 semaphore value = -1
poster_func: Thread 3 posting semaphore
nxsem_restoreholderprio: ERROR: TCB 0x01986290 is a stale handle, counts lost
up_assert: Assertion failed at file:semaphore/sem_holder.c line: 494 task: <pthread>
waiter_func: Thread 2 awakened
waiter_func: Thread 2 new semaphore value = 0
waiter_func: Thread 2 done
```

After it is disabled the semaphore test passes
```
user_main: timed semaphore test
semtimed_test: Initializing semaphore to 0
semtimed_test: Waiting for two second timeout
semtimed_test: PASS: first test returned timeout
BEFORE: (38 sec, 465130108 nsec)
AFTER:  (40 sec, 465288415 nsec)
semtimed_test: Starting poster thread
semtimed_test: Set thread 1 priority to 191
semtimed_test: Starting poster thread 3
semtimed_test: Set thread 3 priority to 64
semtimed_test: Waiting for two second timeout
poster_func: Waiting for 1 second
poster_func: Posting
semtimed_test: PASS: sem_timedwait succeeded
BEFORE: (40 sec, 473380006 nsec)
AFTER:  (41 sec, 476296020 nsec)
```

btashton, 2020-05-03:
When thread 1 is joined by thread 3 its priority is increased to match thread 3 (higher than thread 2) it then exits, but the semaphore still is holding the reference to thread 1 (`*sem.hhead->flink->htcb` is thread 1) so when thread 2 calls sem_post,  `nxsem_restoreholderprio` is called on thread 1 which is stale.

It seems that while we call
`nxsem_boostpriority` in `sem_wait` we should also be calling `nxsem_restorebaseprio` once the task becomes unblocked?

something like this in `sem_wait`
```
#ifdef CONFIG_PRIORITY_INHERITANCE
          nxsem_restorebaseprio(rtcb, sem);
          sched_unlock();
#endif
        }
    }

  leave_critical_section(flags);
  return ret;
}
```

btashton, 2020-05-03:
I also see a few places where the error check has been disabled that were introduced here.  I think the core bit of this is that we are not removing references after we wait, only on the post. 
@patacongo  do you know the history of this one?
https://github.com/apache/incubator-nuttx/commit/7a151016c2b435641884ec41528d1eac5cc79db2

patacongo, 2020-05-03:
I will have to look at the code later, but I think I disagree.  The caller of sem_wait gets to wait at the elevated priority, but once it receives the semaphore, it must NOT be permitted  to run at at that elevated priority.

Therefore the correct place to restore its priority is in sem_post BEFORE  it has a chance to run.  But I will need to check,

patacongo, 2020-05-03:
> I will have to look at the code later, but I think I disagree. The caller of sem_wait gets to wait at the elevated priority, but once it receives the semaphore, it must NOT be permitted to run at at [...]

well, I got that all wrong.  The waiter always waits at its own priority, the thread(s) that hold a semaphore count have their priority boosted..  I shouldn't answer questions while I am still I bed.

In sem_wait(), if the thread was wait, it will boost the execution priority of all of threads that hold a count on the semaphore, then wait at its own priority.

When one of the threads whose priority boost posts the semaphore, then it will restore its execution priority as part of the sem_post.c logic.  nxsem_post() does this:

    177       /* Check if we need to drop the priority of any threads holding
    178        * this semaphore.  The priority could have been boosted while they
    179        * held the semaphore.
    180        */
    181
    182 #ifdef CONFIG_PRIORITY_INHERITANCE
    183       nxsem_restorebaseprio(stcb, sem);
    184       sched_unlock();
    185 #endif

btashton, 2020-05-03:
But sem_post is never called by thread 1 and so when it exits it remains attached to the semaphore which leads to the wrong behavior.

btashton, 2020-05-03:
@patacongo If I place a breakpoint at the end of the waiter function this is what I see on the first hit (thread 1).  The semaphore still had thread 1 in its holder.  I see the priority is returned to what I would expect for the thread, but it should not be attached to the semaphore anymore.
```
Breakpoint 1, waiter_func (parameter=0x1) at sem.c:111
111	  printf("waiter_func: Thread %d done\n",  id);
(gdb) p sem
$1 = {semcount = -1, flags = 0 '\000', hhead = 0x10097bca0 <g_holderalloc>}
(gdb) p *sem->hhead
$2 = {flink = 0x0, htcb = 0x101988290, counts = 1}
(gdb) p *sem->hhead->htcb
$3 = {flink = 0x102188bd0, blink = 0x0, group = 0x101587650, pid = 4, start = 0x10090d639 <pthread_start>, 
  entry = {pthread = 0x10094472e <waiter_func>, main = 0x10094472e <waiter_func>}, 
  sched_priority = 191 '\277', init_priority = 191 '\277', npend_reprio = 0 '\000', 
  pend_reprios = '\000' <repeats 15 times>, base_priority = 191 '\277', task_state = 3 '\003', flags = 1, 
  lockcount = 0, waitdog = 0x0, adj_stack_size = 4194288, stack_alloc_ptr = 0x101988710, 
  adj_stack_ptr = 0x101d886f8, waitsem = 0x0, sigprocmask = 0, sigwaitmask = 0, sigpendactionq = {
    head = 0x0, tail = 0x0}, sigpostedq = {head = 0x0, tail = 0x0}, sigunbinfo = {si_signo = 0 '\000', 
    si_code = 0 '\000', si_errno = 0 '\000', si_value = {sival_int = 0, sival_ptr = 0x0}, si_pid = 0, 
    si_status = 0}, msgwaitq = 0x0, pthread_data = {0x0, 0x0, 0x0, 0x0}, pterrno = 0, xcp = {
    sigdeliver = 0x0, saved_rip = 0, saved_rflags = 0, saved_rsp = 0, regs = {64, 3843995729920, 0, 
      281470681751424, 0 <repeats 63 times>, 16, 16, 1, 0, 4325934624, 0 <repeats 13 times>, 4304610883, 8, 
      6, 4325934592, 16}}, name = "<pthread>", '\000' <repeats 22 times>}
(gdb) 
```

btashton, 2020-05-04:
Ok I think I fully understand what is going on here.  The dangling allocated holder is the one allocated by `sem_post`.

From `sem_post()`
```
          if (stcb != NULL)
            {
              /* The task will be the new holder of the semaphore when
               * it is awakened.
               */
              sinfo("ALLOCATING VIA POST\n");
              nxsem_addholder_tcb(stcb, sem);
```
At this point we have allocated a holder on behalf of thread 1 (tcb 0x101988290)
nxsem_addholder_tcb: INFO: holder 0x10097bca0 allocated with tcb 0x101988290

```
             /* It is, let the task take the semaphore */

              stcb->waitsem = NULL;

              /* Restart the waiting task. */
```
Now we are waking up thread 1.  Here is the issue, this thread is going to exit
```
              up_unblock_task(stcb);
            }
#if 0 /* REVISIT:  This can fire on IOB throttle semaphore */
          else
            {
              /* This should not happen. */

              DEBUGPANIC();
            }
#endif
        }

      /* Check if we need to drop the priority of any threads holding
       * this semaphore.  The priority could have been boosted while they
       * held the semaphore.
       */
```
Now we restore the priority
```
#ifdef CONFIG_PRIORITY_INHERITANCE
      nxsem_restorebaseprio(stcb, sem);
      sched_unlock();
#endif
      ret = OK;
```
But `nxsem_restorebaseprio` is going to panic because the thread has exited
```
  if (!sched_verifytcb(htcb))
    {
      serr("ERROR: TCB %p is a stale handle, counts lost\n", htcb);
      DEBUGPANIC();
      pholder = nxsem_findholder(sem, htcb);
      if (pholder != NULL)
        {
          nxsem_freeholder(sem, pholder);
        }
    }
```

I'm really not sure what we want to do here.  Would it make sense for when a thread exists to remove all of its allocated handlers by iterating through `g_holderalloc`?

patacongo, 2020-05-04:
> I'm really not sure what we want to do here. Would it make sense for when a thread exits to remove all of its allocated handlers by iterating through `g_holderalloc`?

Yes, that is the kind of thing that could be done in the task_exithook().   It would be analogous to sched/signal/sig_cleanup.c .  nxsig_cleanup() is called from task_exithook() to clean up signal-related stuff.

Does nxsem_findholder() and nxsem_freeholder() really require that htcb be valid.  No, they do not appear to.  So I think you could remove the DEBUGPANIC and put a big fat comment in nxsem_findhold() and nxsem_freeholder() that you that htcb must be treated as a look-up key only and should not be de-referenced.  Do you want me to make that change?  You could verify it and and so the merge if it works for you.

patacongo, 2020-05-04:
Please try PR #967 

Based on my understanding of your analysis, it should eliminate the problem.  NOTE that there are other places were the DEBUGPANIC() had to be removed in similar tests.

btashton, 2020-05-04:
> Yes, that is the kind of thing that could be done in the task_exithook().   It would be analogous to sched/signal/sig_cleanup.c .  nxsig_cleanup() is called from task_exithook() to clean up [...]

The downside to not doing the cleanup is the holders are finite and if you have a bunch of threads closing on a semaphore you could exhaust them. Maybe this is something for a future improvement though.

closed 2020-05-04 by btashton