apache/nuttx #955: x86_64 ostest mqueue_test failing

btashton, 2020-05-03:
The test seems to fail in two ways.  With CONFIG_SCHED_CHILD_STATUS enabled
```
user_main: message queue test
mqueue_test: Starting receiver
mqueue_test: Set receiver priority to 128
receiver_thread: Starting
mqueue_test: Starting sender
mqueue_test: Set sender thread priority to 64
mqueue_test: Waiting for sender to complete
sender_thread: Starting
receiver_thread: mq_receive succeeded on msg 0
sender_thread: mq_send succeeded on msg 0
receiver_thread: mq_receive succeeded on msg 1
sender_thread: mq_send succeeded on msg 1
receiver_thread: mq_receive succeeded on msg 2
sender_thread: mq_send succeeded on msg 2
receiver_thread: mq_receive succeeded on msg 3
sender_thread: mq_send succeeded on msg 3
receiver_thread: mq_receive succeeded on msg 4
sender_thread: mq_send succeeded on msg 4
receiver_thread: mq_receive succeeded on msg 5
sender_thread: mq_send succeeded on msg 5
receiver_thread: mq_receive succeeded on msg 6
sender_thread: mq_send succeeded on msg 6
receiver_thread: mq_receive succeeded on msg 7
sender_thread: mq_send succeeded on msg 7
receiver_thread: mq_receive succeeded on msg 8
sender_thread: mq_send succeeded on msg 8
receiver_thread: mq_receive succeeded on msg 9
sender_thread: mq_send succeeded on msg 9
sender_thread: returning nerrors=0
mqueue_test: Killing receiver
up_sigdeliver: Deliver signal to 101982db0: 100914520
pthread_completejoin: ERROR: Could not find join info, pid=42
up_sigdeliver: Deliver signal to 101181190: 100914520
up_assert: Assertion failed at file:sched/sched_waitpid.c line: 554 task: ostest
up_sigdeliver: Deliver signal to 100d7ff90: 100914520
```

and when CONFIG_SCHED_CHILD_STATUS disabled in a different way
```
user_main: message queue test
mqueue_test: Starting receiver
mqueue_test: Set receiver priority to 128
receiver_thread: Starting
mqueue_test: Starting sender
mqueue_test: Set sender thread priority to 64
mqueue_test: Waiting for sender to complete
sender_thread: Starting
receiver_thread: mq_receive succeeded on msg 0
sender_thread: mq_send succeeded on msg 0
receiver_thread: mq_receive succeeded on msg 1
sender_thread: mq_send succeeded on msg 1
receiver_thread: mq_receive succeeded on msg 2
sender_thread: mq_send succeeded on msg 2
receiver_thread: mq_receive succeeded on msg 3
sender_thread: mq_send succeeded on msg 3
receiver_thread: mq_receive succeeded on msg 4
sender_thread: mq_send succeeded on msg 4
receiver_thread: mq_receive succeeded on msg 5
sender_thread: mq_send succeeded on msg 5
receiver_thread: mq_receive succeeded on msg 6
sender_thread: mq_send succeeded on msg 6
receiver_thread: mq_receive succeeded on msg 7
sender_thread: mq_send succeeded on msg 7
receiver_thread: mq_receive succeeded on msg 8
sender_thread: mq_send succeeded on msg 8
receiver_thread: mq_receive succeeded on msg 9
sender_thread: mq_send succeeded on msg 9
sender_thread: returning nerrors=0
mqueue_test: Killing receiver
up_sigdeliver: Deliver signal to 102582b30: 1009130b1
mqueue_test: Canceling receiver
mqueue_test: receiver has already terminated
mqueue_test: ERROR send mqd_t left open
```

sonicyang, 2020-05-06:
@btashton I am also getting this.
I will try look into it.

sonicyang, 2020-05-06:
Well, actually a bit different, mine only appears when set `CONFIG_DEBUG_ASSERTIONS=yes`
It fails with a assert fail in semaphore holder management, during message queue test.
Something related to signaling I think.

sonicyang, 2020-05-06:
Now I am getting another assertion fail at 
```
[   53.749589] up_assert: Assertion failed at file:sched/sched_waitpid.c line: 419 task: init
```

sonicyang, 2020-05-12:
@btashton 
Can you verify for me that whether the bug goes away by disabling CONFIG_SIG_DEFAULT in the ostest config?

I might know the origin of this bug.

btashton, 2020-05-13:
Looks like it passed the mqueue test but then failed the barrier test
```
user_main: barrier test
barrier_test: Initializing barrier
barrier_func: Thread 0 started
barrier_test: Thread 0 created
barrier_func: Thread 1 started
barrier_test: Thread 1 created
barrier_func: Thread 2 started
barrier_test: Thread 2 created
barrier_func: Thread 3 started
barrier_test: Thread 3 created
barrier_func: Thread 4 started
barrier_test: Thread 4 created
barrier_func: Thread 5 started
barrier_test: Thread 5 created
barrier_func: Thread 6 started
barrier_test: Thread 6 created
barrier_func: Thread 7 started
barrier_test: Thread 7 created
barrier_func: Thread 0 calling pthread_barrier_wait()
barrier_func: Thread 1 calling pthread_barrier_wait()
barrier_func: Thread 2 calling pthread_barrier_wait()
barrier_func: Thread 3 calling pthread_barrier_wait()
barrier_func: Thread 4 calling pthread_barrier_wait()
barrier_func: Thread 5 calling pthread_barrier_wait()
barrier_func: Thread 6 calling pthread_barrier_wait()
barrier_func: Thread 7 calling pthread_barrier_wait()
barrier_func: Thread 7, back with status=PTHREAD_BARRIER_SERIAL_THREAD (I AM SPECIAL)
barrier_func: Thread 0, back with status=0 (I am not special)
barrier_func: Thread 1, back with status=0 (I am not special)
barrier_func: Thread 2, back with status=0 (I am not special)
barrier_func: Thread 3, back with status=0 (I am not special)
barrier_func: Thread 4, back with status=0 (I am not special)
barrier_func: Thread 5, back with status=0 (I am not special)
barrier_func: Thread 6, back with status=0 (I am not special)
barrier_func: Thread 7 done
barrier_func: Thread 0 done
barrier_func: Thread 1 done
barrier_func: Thread 2 done
barrier_func: Thread 3 done
barrier_func: Thread 4 done
barrier_test: Thread 0 completed with result=0
barrier_func: Thread 5 done
barrier_func: Thread 6 done
barrier_test: Thread 1 completed with result=0
barrier_test: Thread 2 completed with result=0
barrier_test: Thread 3 completed with result=0
barrier_test: Thread 4 completed with result=0
barrier_test: Thread 5 completed with result=0
barrier_test: Thread 6 completed with result=0
barrier_test: Thread 7 completed with result=0
up_assert: Assertion failed at file:semaphore/sem_holder.c line: 908 task: ostest
ostest_main: Exiting with status 256
```

sonicyang, 2020-05-13:
@btashton I am also getting this,. After I further disabling the priority inheritance, everything works fine. (In terms, this also enabled some other test which are not priority inheritance compatible.)

I am curious, are there any suggested ostest configuration?
I do think we found some corner cases.

btashton, 2020-05-13:
I found another bug related to priority inheritance running the ostest for x86_64 (well really just faulty logic in the debug assert) that we closed.  So there might be another one, it gets a little hard to reason through, which is why I just opened the bug for when I (or someone else) feels inspired to dig into it.  I have not noticed any issues caused by it yet.

sonicyang, 2020-05-13:
I opened a issue about mqueue_test which is the root cause of the initial problem of this issue.