apache/nuttx #956: Is z80 compiler still not support CONFIG_CPP_HAVE_VARARGS?

xiaoxiang781216, 2020-05-03:
I saw that log related code check CONFIG_CPP_HAVE_VARARGS, but other function e.g. mq_open/ioctl/fcntl/printf doesn't. I am curious how z80 can pass the compile.

patacongo, 2020-05-03:
The refers only VARARGS are used in preprocessor macros.  Like in debug.h:

     104 #ifdef CONFIG_CPP_HAVE_VARARGS
     138 #  define _info(format, ...) \
     139    __arch_syslog(LOG_INFO, EXTRA_FMT format EXTRA_ARG, ##__VA_ARGS__)
     722 #else /* CONFIG_CPP_HAVE_VARARGS */
     739 #  define _info       (void)
    1320 #endif /* CONFIG_CPP_HAVE_VARARGS */

This has nothing to do with the compiler.  It is a feature of the C preprocessor.  This is not a problem for the SDCC toolchain used with z80 and z180 because they use the same GNU pre-processor as does GCC.  However, it is a problem for the ZDS-II toolchain which use a different pre-processor and cannot handled pre-processor macros with a variable number of arguments.

The setting CONFIG_CPP_HAVE_VARARGS only applies to the pre-processor and has nothing to do with variadic C functions with is part of the C standard and is handled by all C compiler.

xiaoxiang781216, 2020-05-03:
Thanks, got the CONFIG_CPP_HAVE_VARARGS meaning.

patacongo, 2020-05-03:
There are other issues with z80 and z180:  I have not verified that they build correctly with the latest configurations that always assume we can pass structures as arguments.  That is supposed to work with the lastest SDCC toolchain, but I have not yet tried.  I added this warning to the README file for both of these architectures:

    IMPORTANT NOTE as of 2020-4-11:  Support for CONFIG_CAN_PASS_STRUCTS was
    removed in NuttX-9.0.  This was necessary to enforce some POSIX interface
    compliance but also means that ALL older SDCC versions will no long build
    with NuttX.  I have been told that the newest SDCC compilers can indeed
    pass structure and union parameters and return values.  If that is correct,
    then perhaps the newer SDCC compilers will be used.  Otherwise, it will be
    necessary to use some other, more compliant compiler.

I think that z80 is no longer a viable platform because of the 16-bit address space.  But the other variants are still good options for retro computing people includin the z180 which has an MMU and the z380 which can support up to a 32-bit address space.  Also the new newer ez80 which handles a 24-bit address space (but does not use SDCC).

closed 2020-05-04 by xiaoxiang781216