apache/nuttx #994: Naming of OS Internal Functions

patacongo, 2020-05-07:
There has been some controvery lately about the consistency of naming of OS internal functions.  We should come to a consensus on the desired naming of conventions for internal OS functions before we start arbitrarily changing names.

NuttX function names are all lower case.  That is required by the naming standard.  Historically, has used a subsystem name, an underscore, then all of the renaming words smooshed together.  For example.

    void sched_mergeprioritized(FAR dq_queue_t *list1, FAR dq_queue_t *list2,
                                uint8_t task_state);

I would represent that form as:

    <subsystem>_<everythingelsesmooshedaltogether>

Many of these names are impossible to read unless you stop and parse the letters to find the word boundaries.

Most recent naming has broken out one more of the smooshed-together-words and separated  them with undersore characters like:

    unsigned int sched_timer_cancel(void);
    void sched_timer_resume(void);
    void sched_timer_reassess(void);

Which I would describe as:

    <subsystem>_<object>_<verb>

Another form, which I have been using lately, is to switch the `<object>` and the `<verb>` like:

    <subsystem>_<verb>_<object>

For example:

    int sched_get_stackinfo(pid_t pid, FAR struct stackinfo_s *stackinfo);

By the `<subsystem>_<object>_<verb>` naming this would have been:

    int sched_stackinfo_get(pid_t pid, FAR struct stackinfo_s *stackinfo);

And the timer interfaces would become the following under the `<subsystem>_<verb>_<object>` rule:

    unsigned int sched_cancel_timerl(void);
    void sched_resume_timer(void);
    void sched_reassess_timer(void);

I won't even consider the `<subsystem>_<everythingelsesmooshedaltogether>`.  It has a history but is pretty much odious

There is other naming that uses long names with many underscore characters separating each word in the long name.  I am personally not a fan of long names.

My preference would be the `<subsystem>_<verb>_<object>`  form.  I have been using this form in creating all new internal interfaces.  Does anyone else have a thought?  or a preference?  We should come to an understand and stop the arbitrary name changes.

NOTE:  This applies only to the internal naming of functions within the OS.  Other names are imposed on us by standards for application interfaces to the OS which may follow other rules.

xiaoxiang781216, 2020-05-07:
Any rule is fine, the most important thing is try the best to follow the rule once the community make the decision. The consistence is much more important than any rule. Wny do we spend the time define a set of rule? the most reason is that we want the code base keep the consistence.

patacongo, 2020-05-07:
> Any rule is fine, the most important thing is try the best to follow the rule once the community make the decision. The consistence is much more important than any rule. Why do we spend the time define a set of rule? Not the rule self, the most reason is that we want the code base keep the consistence.

Yes, for consistency.  But also so that we can work together with constantly modifying other people's work to make it conform to what we believe the way things should be named.  We have to stop doing that with no justification.

xiaoxiang781216, 2020-05-07:
Once the rule is lockdown, the rule is the justification, we don't need more.

patacongo, 2020-05-08:
Discussion moved to https://cwiki.apache.org/confluence/display/NUTTX/Naming+of+OS+Internal+Functions

closed 2020-05-08 by patacongo