apache/nuttx #1010: tools/refresh.sh --help

patacongo, 2020-05-08:
The help text output from `tools/refresh.sh --help` do not see to correctly described the current behavior of this tool.  I imaging the tools/README.txt should t updated as well(?)

patacongo, 2020-05-08:
This reason for this is:

    if ! $CMPCONFIG $DEFCONFIG defconfig; then
    ...

So if no logical content of the defconfig has changed then it will not be updated.  That means if the ordering of elements changes so that there is a lexical content changes, it will not be updated and the user will not be prompted.

Is that good?  It is certainly not as bad as my original perception.

patacongo, 2020-05-10:
I'm closing this issue.  I don't think that there is any real problem that needs fixed.  There is some change in functionality which is probably not desirable, but neither is it a bug or a problem.

xiaoxiang781216, 2020-05-11:
> This reason for this is:
>
> ```
> if ! $CMPCONFIG $DEFCONFIG defconfig; then
> ...
> ```
>
> So if no logical content of the defconfig has changed then it will not be updated. That means if the ordering of [...]

But I think the root cause is that cmpconfig ignore the order difference. We should replace with diff command instead.

> 
> Is that good? It is certainly not as bad as my original perception.

No, it isn't good, I will fix this problem.

patacongo, 2020-05-11:
> But I think the root cause is that cmpconfig ignore the order difference. We should replace with diff command instead.

The use case where I experienced this was as follows:

1. I use find and sed to modify a configuration setting name.  This is a dangerous practice because it can create an invalid configuration.

2. I use refresh.sh to fix this:  It will create a valid defconfig and, if the change was innocuous, it will not show any differences.  However, it may result in a re-ordering of defconfig setting, putting them back in alphabetic order.

3. In either case, you would like to see what is different and be able to save the updated defconfig with the configs in the correct order.  The old version of refresh.sh used tools/compconfig and it might show no differences in this case.  However, it still allowed the opportunity to save the defconfig file, even if there were no differences.

xiaoxiang781216, 2020-05-11:
> The use case where I experienced this was as follows:
>
> 1. I use find and sed to modify a configuration setting name.  This is a dangerous practice because it can create an invalid configuration.
> 2. [...]

Yes, the real difference is that the order change will show to the end user now. The behaviour change is minor, but simplify how we implment the github action.

closed 2020-05-11 by patacongo