apache/nuttx #1031: nxstyle: reports brace alignment problem on solitary blocks

v01d, 2020-05-12:
When having something like:
<pre>
void function(void)
{
  .. code ..
  {
    ... code ...
  }
}
</pre>
it reports a problem on the inner braces as if they were not aligned correctly.

patacongo, 2020-05-12:
nxstyle expects all braces to align in vertical.  Indenting by two spaces vs. the usual for causes nxstyle to complete.  The coding standard is silent on this but does say that all of the braces should align vertically.

When I encounter this complaint from nxstyle, I change the code as follows:

    void function(void)
    {
      .. code ..
        {
          ... code ...
        }
    }

And nxstyle is appy.

v01d, 2020-05-12:
I imagined that this is what nxstyle expects, but as you say the standard does not define this. Maybe if that is the expected style the standard should be augmented.