apache/nuttx #1087: ARMv8-A support in NuttX

patacongo, 2020-05-20:
This was Issue 92 in the old Bitbucket NuttX repository

Frank Wang
created an issue 2018-04-12

Hi Gregory,

Recently, I tried porting NuttX on the Rock-chip platform and found it is one excellent RTOS.

I notice that NuttX have already supported ARMv7-A architecture for a long time, but the ARMv8 still did not involve in current.

As you know, the hardware is updated very quickly from Moore's law, and the customer become more sense on the speed of electronic equipment, so maybe it is a trend that the RTOS can run on the multi-core CPU or the higher performance CPU like 64 bits.

So I would like to know do you have a plan for supporting ARMv8 architecture in NuttX or already have some temporary patches? I will be glad to test them on Rock-chip platform.

Look forward to your reply and thanks for your time.

BR. Frank

Gregory Nutt
2018-04-12

Yes, I would hope to support ARMv8-A (as well as the other ARMv8's) at some point. There simply has been no motivation for that support you since there is no board using any ARMv8-x in the source tree now. That support would be added at the time that there is need.

I don't really believe that NuttX is a good match for the higher end ARMs. There are a few ports out there for Cortex-A8 and A9 but, frankly, no one uses them. SMP and TrustZone on the i.MX6 quad was part of a research project with a University but otherwise it is never used.

I have heard of people using NuttX as TrustZone masters on high end products but, otherwise, as far as I know, those part are not used with NuttX. So to me it is a waste of time porting NuttX to high end "application" processors that were clearly designed to support Linux or Android.

But there are a number of cross-over SoC's where NuttX has been very successful: The SAMA52/3/4 for example. A scaled down Cortex-A5 for embedded use. NuttX is a great match there. The SAMA5Dx is an embedded MCU with a scaled down ARMv7-A core.

The new i.MX RT is another case of a crossover SoC. That port is under development now and I expect it to also be successful. That one is a high end i.MX MCU with a scaled down MCU (300MHz Cortex-M7).

Greg

Kejun ZHOU
2019-04-21

I am porting Nuttx to RaspberryPi3B+ board running in AArch64 mode. Some foundational code is ready in my git https://zhoukejun@bitbucket.org/zhoukejun/nuttx_aarch64_raspberrypi3bplus.git .

A lot of work is needed to be done.
        
Frank Wang
2019-04-25

Althouth, I haven't followed it for a while, but it's still a good news! Keep moving :-)

masayuki2009, 2023-01-13:
https://github.com/apache/nuttx/pull/6478

masayuki2009, 2023-01-13:
nuttx-10.4 supports arm64 architecture

closed 2023-01-13 by masayuki2009