apache/nuttx #1091: features.h:  Out of date Clang support

patacongo, 2020-05-20:
This was Issue 162 in the old Bitbucket repository

Jon Taylor
created an issue 2019-08-26

Hi,

In nuttx/include/features.h, there is a missing block found in my /usr/include/features.h:

    /* Similarly for clang. Features added to GCC after version 4.2 may
    or may not also be available in clang, and clang's definitions of
    __GNUC(_MINOR)__ are fixed at 4 and 2 respectively. Not all such
    features can be queried via __has_extension/__has_feature. */
    #if defined __clang_major__ && defined __clang_minor__
    # define __glibc_clang_prereq(maj, min) \
    ((__clang_major__ << 16) + __clang_minor__ >= ((maj) << 16) + (min))
    #else
    # define __glibc_clang_prereq(maj, min) 0
    #endif

…which leads to build problems with NuttX on my Ubuntu 19 box. The features.h file in general appears to be a copy of glibc’s features.h, which might therefore be out of date and need to be updated…?

yamt, 2020-05-22:
i'm not sure what's "nuttx/include/features.h".

xiaoxiang781216, 2020-07-17:
NuttX don't have include/features.h, let's close this issue.

closed 2020-07-17 by xiaoxiang781216