apache/nuttx #1114: testbuild.sh:  Fails if EXTRA_FLAGS not defned

patacongo, 2020-05-24:
`makefunc` causes `testbuild.sh` to fail if `EXTRA_FLAGS` is not defined on the command line with the `-e `option:

    function makefunc {
      if ! ${MAKE} ${MAKE_FLAGS} "_${EXTRA_FLAGS}"_ $@ 1>/dev/null; then
        fail=1
      fi
    }

Results in in, for example (ALL make commands fail n this case):

    + makefunc olddefconfig
    + make -k '' olddefconfig
    make: *** empty string invalid as file name.  Stop.

This is a consequence of the quotation marks around $(EXTRA_FLAGS) which appear on the command line if EXTRA_FLAGS is not defined.

patacongo, 2020-05-24:
Perhaps something like this would do the job:

    diff --git a/tools/testbuild.sh b/tools/testbuild.sh
    index 91efb5354e..6895ef181e 100755
    --- a/tools/testbuild.sh
    +++ b/tools/testbuild.sh
    @@ -89,7 +89,7 @@ while [ ! -z "$1" ]; do
         ;;
       -e )
         shift
    -    EXTRA_FLAGS="EXTRAFLAGS=$1"
    +    EXTRA_FLAGS="EXTRAFLAGS=\"$1\""
         ;;
       -x )
         MAKE_FLAGS='--silent --no-print-directory'
    @@ -158,7 +158,7 @@ blacklist=`grep "^-" $testfile || true`
     cd $nuttx || { echo "ERROR: failed to CD to $nuttx"; exit 1; }
    
     function makefunc {
    -  if ! ${MAKE} ${MAKE_FLAGS} "${EXTRA_FLAGS}" $@ 1>/dev/null; then
    +  if ! ${MAKE} ${MAKE_FLAGS} ${EXTRA_FLAGS} $@ 1>/dev/null; then
         fail=1
       fi
     }

The backslash will retain the " inside of the EXTRA_FLAGS string and it should be present on the command line when ${EXTRA_FLAGS} is expanded.

liuguo09, 2020-05-25:
If call testbuild.sh with the change you made, it works well with  _-e "-Wno-cpp"_ is OK, but it doesn't work with  _-e "-Wno-cpp -Werror"_ , fail log as below. 

_+ make --silent --no-print-directory 'EXTRAFLAGS="-Wno-cpp' '-Werror"' -j 4
 /bin/sh: 1: Syntax error: Unterminated quoted string_

We may need find another way to fix both qeuestions.

> Perhaps something like this would do the job:
> 
> ```
> diff --git a/tools/testbuild.sh b/tools/testbuild.sh
> index 91efb5354e..6895ef181e 100755
> --- a/tools/testbuild.sh
> +++ b/tools/testbuild.sh
> @@ -89,7 +89,7 @@ while [ ! -z "$1" ]; do
>      ;;
>    -e )
>      shift
> -    EXTRA_FLAGS="EXTRAFLAGS=$1"
> +    EXTRA_FLAGS="EXTRAFLAGS=\"$1\""
>      ;;
>    -x )
>      MAKE_FLAGS='--silent --no-print-directory'
> @@ -158,7 +158,7 @@ blacklist=`grep "^-" $testfile || true`
>  cd $nuttx || { echo "ERROR: failed to CD to $nuttx"; exit 1; }
> 
>  function makefunc {
> -  if ! ${MAKE} ${MAKE_FLAGS} "${EXTRA_FLAGS}" $@ 1>/dev/null; then
> +  if ! ${MAKE} ${MAKE_FLAGS} ${EXTRA_FLAGS} $@ 1>/dev/null; then
>      fail=1
>    fi
>  }
> ```
> 
> The backslash will retain the " inside of the EXTRA_FLAGS string and it should be present on the command line when ${EXTRA_FLAGS} is expanded. If EXTRA_FLAGS is not defined then nothing will be put on the command line.

closed 2020-05-26 by patacongo