apache/nuttx #1131: Common Driver Initialization

patacongo, 2020-05-26:
There are several implementations of `up_initialize()`, at least one per architecture.  Each of these initializtion files is very similar; each initializes the same set of common drivers under the `drivers/` directory.

This common logic should be captured in common logic perhaps at `/drivers/init`.  The common driver initialization should be removed from architecture-specific `up_initialize()` and replaces with calls to `drvr_early_initialize()` and` drvr_last_initialize()` in new files under `/drivers/init`.

dalprahcd, 2024-02-03:
Hello @patacongo, is this still a valid issue? If so, I would like to try helping with.

xiaoxiang781216, 2024-02-16:
@dalprahcd some suggestion was done recently, you can reference the code here:
https://github.com/apache/nuttx/blob/master/drivers/drivers_initialize.c

dalprahcd, 2024-02-17:
@xiaoxiang781216 since there is a common file to initialize all the drivers, is this issue closed then? Because checking the following architectures looks like there is no much duplication at `up_intialize`.

https://github.com/apache/nuttx/blob/42dad89211f2d34f62c103a7647bd64e466a0215/arch/avr/src/common/avr_initialize.c#L125
https://github.com/apache/nuttx/blob/42dad89211f2d34f62c103a7647bd64e466a0215/arch/arm/src/common/arm_initialize.c#L98
https://github.com/apache/nuttx/blob/42dad89211f2d34f62c103a7647bd64e466a0215/arch/risc-v/src/common/riscv_initialize.c#L88
https://github.com/apache/nuttx/blob/42dad89211f2d34f62c103a7647bd64e466a0215/arch/xtensa/src/common/xtensa_initialize.c#L101

xiaoxiang781216, 2024-11-01:
Fixed by https://github.com/apache/nuttx/pull/5728

xiaoxiang781216, 2024-11-01:
> @xiaoxiang781216 since there is a common file to initialize all the drivers, is this issue closed then? Because checking the following architectures looks like there is no much duplication at [...]

sorry for the late rely, I miss it from my email. Yes, I think so.

closed 2024-11-01 by xiaoxiang781216