apache/nuttx #1138: Thread can be suspended within a critical section.

patacongo, 2020-05-27:
Occasionally, you will see an odd behavior while within a critical section:  A thread may be unexpectedly suspended.

The proposed solution to this problem is to lock the scheduler while within a critical section.

This solution, however, depends first on resolution of PR #1137 

@xiaoxiang781216 Can you clarify the circumstance under which this was recurring.  My old, tired brain cannot recall the details.  This problem was originally reported by Guiding Li.  As I recall there were two scenarios where unexpected context switches occurred:  One involved PM state changes and the other involved cancellation of wdog timer.  It would be good to get those two failure scenarios detailed here.

xiaoxiang781216, 2020-05-27:
@liguiding please take took.

xiaoxiang781216, 2020-05-27:
@patacongo another issue related to crtical section: crtical section should be replaced with semphare or spinlock as much as we can to improve SMP performance.

closed 2025-01-23 by xiaoxiang781216