apache/nuttx #1158: spresense:wifi multiple definition of g_video_devops

patacongo, 2020-05-31:
This error occurs because:

1. `g_video_devops` is declared in `video.h` with no `extern`.  So every inclusion of `video.h` creates a new instance of `g_video_devops`.  Hence, the "multiple definition" error at link time.

    video/video.h:FAR const struct video_devops_s *g_video_devops;

2. There is no other declaration of` g_video_devops` (at least as far as I can find).  If you correct the problem by adding the `extern` to the declaration in `video.h`, then I think you still get a link time error because `g_video_devops` would be undefined.

Use of global variables in **_extremely_** bad style in an RTOS.  The correct fix is to remove `g_video_devops` altogether and pass the instance as a parameter as is done fore every other driver.  Related Issue:  #439

patacongo, 2020-05-31:
I also ran the complete ARM build tests locally here.  Of the 436 ARM configurations in that test, only the spresense:wifi build failed.  However, it failed with a different error:

udpblaster_host.c:40:10: fatal error: config.h: No such file

masayuki2009, 2020-05-31:
Hmm, I can not reproduce the build errors for spresense:wifi.
I did build tests on my ubuntu machine.

However, we should fix g_video_devops in video.h. 
Hi @jerpelea, please take a look at the code.

masayuki2009, 2020-06-01:
> I also ran the complete ARM build tests locally here. Of the 436 ARM configurations in that test, only the spresense:wifi build failed. However, it failed with a different error:
> 
> udpblaster_host.c:40:10: fatal error: config.h: No such file

Hi @patacongo,

I can not reproduce this build error with my ubuntu machine.
And it seems strange to me, because other configurations such as sama5d2-xult:netnsh also include CONFIG_EXAMPLES_UDPBLASTER. So I'm still not sure why the above error happens only for spresense:wifi.

Ouss4, 2020-06-01:
> Hmm, I can not reproduce the build errors for spresense:wifi.

Not even the linker's "multiple definition" errors?
I have a clean compilation stage (no udpblaster error) but then the linker fails with that error.

patacongo, 2020-06-01:
> I can not reproduce this build error with my ubuntu machine.
> And it seems strange to me, because other configurations such as sama5d2-xult:netnsh also include CONFIG_EXAMPLES_UDPBLASTER. So I'm still not sure why the above error happens only for spresense:wifi.

I didn't look into the issue in detail.  I think some of what we are seeing may be toolchain-related differences.  I cannot be sure of that.  There have been other cases of the past of global variables defined in header files that were there for years before someone got the error.

The global variable in video.h is an issue, whether it generating failures or not.

I did not see that problem.  I saw a problem finding config.h.  The config.h problem could also be an include path issue with the toolchain.   apps/examples/udpblaster copies the config.h header file as a local config.h.  That is what it has having trouble finding.

Which toolchain are you using? @Ouss4 which toolchain are you using?  I found the problem with the current ARM Embedded toolchain.

I will look into this more tomorrow.  It is late here.

Ouss4, 2020-06-01:
I'm also using the ARM GCC toolchain.

patacongo, 2020-06-01:
A little web research suggests that most GCC compilers will not complain about multiple global definitions in header files unless the globals are initialized, which they are not here.

masayuki2009, 2020-06-01:
I've been using arm gcc 9.2.1

$ arm-none-eabi-gcc --version
arm-none-eabi-gcc (GNU Tools for Arm Embedded Processors 9-2019-q4-major) 9.2.1 20191025 (release) [ARM/arm-9-branch revision 277599]
Copyright (C) 2019 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

closed 2020-06-01 by Ouss4

patacongo, 2020-06-02:
@Ouss4 has verified that the change of PR #1170 does eliminate his build problems.

> udpblaster_host.c:40:10: fatal error: config.h: No such file

@xiaoxiang781216  I also could not reproduce the problem that I saw with the missing config.h file.  I am thinking that this was a one-time parallel build problem and unrelated to this issue.  Xiang, coul you look at the Makefile in apps/examples/udbblaster to see if you can see in obvious race condition?  You have a good eye for those.

xiaoxiang781216, 2020-06-02:
@papatience @Ouss4 it seems that the dependence in Makefile is wrong, please review the fix at:
https://github.com/apache/incubator-nuttx-apps/pull/283