apache/nuttx #1235: nxstyle should not complain on _Exit

yamt, 2020-06-15:
e.g.

    % nxstyle stdlib.h
    stdlib.h:152:19: error: Mixed case identifier found

v01d, 2020-06-16:
Maybe nxstyle could be extended to recognize some special comment that instructs to ignore a line (or at least a given rule for that line)

patacongo, 2020-06-16:
> 
> 
> Maybe nxstyle could be extended to recognize some special comment that instructs to ignore a line (or at least a given rule for that line)

This has been discussed and I, for one, have been very opposed to corrupting good, clean C files with meta-data for nxstyle.  People have talked about a lot of things and we have done the best we can to squash that approach.

yamt, 2020-06-17:
for this specific case (_Exit), as it's a part of the standards, it's more appropriate to make nxstyle itself know, IMO.

Ouss4, 2020-06-18:
There already is a white list in nxstyle.c, we can just add `_Exit` to the list.
```diff
diff --git a/tools/nxstyle.c b/tools/nxstyle.c
index 78351334bf..38b89f98df 100644
--- a/tools/nxstyle.c
+++ b/tools/nxstyle.c
@@ -192,6 +192,7 @@ static const char *g_white_prefix[] =
   "ub8",     /* Ref:  include/fixedmath.h */
   "ub16",    /* Ref:  include/fixedmath.h */
   "ub32",    /* Ref:  include/fixedmath.h */
+  "_Exit",   /* Ref:  include/stdlib.h */
   NULL
 };
```

closed 2020-06-28 by Ouss4