apache/nuttx #1284: zipme.sh is broken for creating release tarballs

btashton, 2020-06-23:
The tool broke some time after the 9.0.0 release.  It looks like a change was made to how we handled versions and making it optional, but it cannot function as before for creating the release.

Also it does not have the execute bit set which should also be fixed.

```
❯ ./zipme.sh -d -s -k 9750ED7E692B99E2 9.1.0
+ DEBUG=-d
+ shift
+ '[' '!' -z -s ']'
+ case $1 in
+ sign=1
+ shift
+ '[' '!' -z -k ']'
+ case $1 in
+ shift
+ GPG+=' --default-key 9750ED7E692B99E2'
+ shift
+ '[' '!' -z 9.1.0 ']'
+ case $1 in
+ break
+ VERSION=9.1.0
./zipme.sh: line 338: syntax error: unexpected end of file
```

btashton, 2020-06-23:
@xiaoxiang781216 I think you made the version changes can you take a look?  I went ahead and created the release tarball using the tool from 9.0 outside of the repo, so it is not blocking the release process but we should get this fixed for the next release.

xiaoxiang781216, 2020-06-23:
Please try this: https://github.com/apache/incubator-nuttx/pull/1285

closed 2020-06-23 by btashton