apache/nuttx #1329: System calls require address checks

patacongo, 2020-06-29:
As a security measure, system calls such as read() or any other system call that modifies memory, must verify that the user-provided addresses lie in user-space.  Otherwise, this opens a back door that may be exploited to take control of the OS or to crash the OS.

This applies only to PROTECTED and KERNEL build modes.

patacongo, 2020-07-03:
Related to Issue #1359

btashton, 2023-04-26:
For myself and others reference later there is some information on how the Linux kernel handles this here 
https://linux-kernel-labs.github.io/refs/pull/187/merge/lectures/syscalls.html#system-call-parameters-handling

Some notes on further hardened usercopy in Linux https://lwn.net/Articles/693745/

From freebsd the copyin copyout framework 
https://freebsdfoundation.org/wp-content/uploads/2021/07/The-copyinout-Framework.pdf

casaroli, 2026-10-19:
I reviewed and tested the four open PRs of @royzah that address this issue, and I want to keep the points that concern all of them in one place.

| PR | What it does | State |
|---|---|---|
| #20412 | Helpers that refuse kernel addresses from user space in a kernel build (`uaccess_ok()`, `uaccess_check()`, `uaccess_fault()`), page-table walkers for riscv and x86_64 | Open; nothing calls the checks yet, the others use them |
| #20418 | Copies in the ioctls and calls that carry user pointers, and refuses kernel pointers | Draft, on top of #20412 |
| #20415 | Keeps user semaphores in the kernel | Draft, on top of #20412 |
| #20425 | Signalling or rescheduling another process needs `PR_CAP_ADMIN` | Draft; independent of #20412 (#20417 is merged) |

All four rebase cleanly on master (c7cba680a10), build, and pass ostest on `qemu-armv8a:knsh` and `:nsh`. My reviews: https://github.com/apache/nuttx/pull/20412#pullrequestreview-5575023201, https://github.com/apache/nuttx/pull/20418#pullrequestreview-5575023672, https://github.com/apache/nuttx/pull/20415#pullrequestreview-5575024174, https://github.com/apache/nuttx/pull/20425#pullrequestreview-5575024560.

**Merge order.** #20412 first: it is the base and it is closest to merge. Then #20418 and #20415, each as one commit on top of it. #20425 can go at any time.

**Open design point.** #20418 (and #20415) still generate per-call checks in `tools/mksyscall.c`, which xiaoxiang asked to keep out of the general stub/proxy creator in the #20412 review. #20425 puts its checks into the functions instead. Can we agree on one place before the next round?

**Gap that none of the four closes yet.** The checks accept any address inside the user window, also an address that is not mapped. Then the kernel faults on it at EL1, and on arm64 a kernel fault is a panic. So any program in a kernel build can stop the system:

```c
read(fd, (void *)0xF0000000, 1); /* in the user window, not mapped */
```

On master, and with all four PRs, this panics the kernel (`arm64_fatal.c:696`). A kernel pointer (`0x1000`) is refused with #20418, which is good. To close the gap, the kernel must recover from a fault on a user address during a system call: either a fixup table for the user-access code (as `copy_from_user()` does on Linux), or a rule that a fault at EL1 on a user address in a system call ends the caller instead of the system.

**Small, older point.** `ARCH_ADDRENV_VEND` and `ARCH_SHM_VEND` are inclusive, but `arm64_uservaddr()`, `riscv_uservaddr()`, the armv7-a copy and the new x86_64 `up_addrenv_user_vaddr()` test `vaddr < VEND`, so the last byte of each window counts as "not user". It is older than these PRs; a small separate cleanup would fix it.

I am happy to test new versions.

patacongo, 2026-10-19:
Yes, certainly we must never use pointers from user space within the kernel.  Nor should be ever attempt access other user stuff in the kernel like semaphores, file descriptors, anything that is specific to a task group.

It won't work in kernel mode

That would be the moral equivalent of accessing user process data directly directly in the Linux kernel (without going through any special Linux user space access functions.

This id because a context switch can occur at any time making the user data inaccessible or worse.

The typical fix is to move things like user data, semaphores, etc. into TLS and not carry them inside the kernel.  TLS resides in the task group user space and doesn't cause a problems like this.

#20412<https://github.com/apache/nuttx/pull/20412>
#20418<https://github.com/apache/nuttx/pull/20418>
#20415<https://github.com/apache/nuttx/pull/20415>
#20425<https://github.com/apache/nuttx/pull/20425>

), build, and pass ostest on qemu-armv8a:knsh and :nsh. My reviews: #20412 (review)<https://github.com/apache/nuttx/pull/20412#pullrequestreview-5575023201>, #20418 (review)<https://github.com/apache/nuttx/pull/20418#pullrequestreview-5575023672>, #20415 (review)<https://github.com/apache/nuttx/pull/20415#pullrequestreview-5575024174>, #20425 (review)<https://github.com/apache/nuttx/pull/20425#pullrequestreview-5575024560>.

casaroli, 2026-10-19:
A note on how a fault on a user address fits the current series. #20412 and #20418 check that a user pointer is in the user address window, and end the caller when it is a kernel address. A pointer in the window that has no page passes that check, for example `read(fd, (void *)0xf0000000, 1)` on qemu-armv8a:knsh. The kernel then faults at EL1 when it touches the pointer, and arm64 panics.

#20822 is an arm64 change for that case, as a fallback: when the kernel takes a translation, access flag or permission fault at EL1 on a user address during a system call of a user task, the handler ends the caller with `_exit(SIGSEGV)`, the same result that `uaccess_fault()` gives for a kernel address. Faults on kernel addresses still panic. It touches `arch/arm64` (the fault handler and `dispatch_syscall()`), two labels in `libs/libc/machine/arm64/arch_memcpy.S`/`arch_memset.S`, and in two more commits `fs/vfs` (the stat family and `nx_vioctl()`), so it does not overlap the files of the series.

Result on qemu-armv8a:knsh (HVF) with a test program that gives 11 system calls a bad user pointer, each in a forked child (read, write to a pipe, stat, ioctl, nanosleep, clock_gettime, sigprocmask, a read into program text, a read that runs from a mapped page into an unmapped page):

- master: kernel panic on the first case (`common/arm64_fatal.c:696`).
- with the change: each child ends with exit status 11 and NSH continues, 231 of 231 over 21 passes; ostest exits 0 on knsh and nsh.
- kernel heap: no growth for 20 faulting children of any case (also `fstat()`, `statfs()` and `fstatfs()` with a bad buffer).

A fault in the arm64 `memcpy()` or `memset()` returns from that function, so the system call finishes and releases its locks and references; the caller ends when the system call returns. The stat family now writes the user buffer with one `memcpy()` after it releases its references, and `ioctl(FIONBIO)` reads its `int` before it takes the file reference. FIONBIO is not one of the ioctls that #20418 copies in, so the two do not overlap. The limit: a fault in other code stops the system call in the middle, so a lock or a reference that it holds stays as it is, and the caller gets no `EFAULT`. A user-access layer with fault fixups (`copy_from_user()`-style helpers that return `-EFAULT`) can come later, path by path, starting with the #20418 wrappers. The kill path then stays as the fallback for code that touches user memory directly.

royzah, 2026-10-24:
Thanks @casaroli for testing these and opening https://github.com/apache/nuttx/pull/20822.

Yeah, the window check isnt enough. Lets keep #20412 as the base, then move the per-call checks out of `mksyscall.c` in #20418 and check full buffers.

For #20415, I'll check the IRQ fallback and semaphore overhead first. I'll rebase #20425 now that #20417 landed.

Your arm64 fallback helps, but we still need safe user copies returning `EFAULT` with proper cleanup.