apache/nuttx #1354: Bug in can driver: wrong use of nxsem_get_value

lukegluke, 2020-07-02:
nxsem_get_value is absolutely wrongly used in can.c
https://github.com/apache/incubator-nuttx/blob/57bc329aace6d816b9a51a368665ea4c7116a9c6/drivers/can/can.c#L1346

it should be
`if (nxsem_get_value(&fifo->rx_sem, &sval) == 0 && sval <= 0)`

or even better to check first one separately with debug assert and optional error output, if CONFIG_CAN_ERRORS defined, like in other drivers.

This bug leads to irreversible increase of fifo->rx_sem with each received msg until finally get into `DEBUGASSERT(sem->semcount<SEM_VALUE_MAX);` in sem_post.c. So in fact without debug can driver get broken after 32767 messages - stopping receiving anything at all (while tx is still working).
I wonder why nobody do not encountered this issue yet? 

I'm not good in kernel stuff, as for me, while digging this it was hard to understand description for nxsem_get_value [here ](https://github.com/apache/incubator-nuttx/blob/57bc329aace6d816b9a51a368665ea4c7116a9c6/libs/libc/semaphore/sem_getvalue.c#L62)
```
If 'sem' is locked, the value return by nxsem_get_value() will either be zero or a negative number whose absolute value represents the number of tasks waiting for the semaphore.
```
It can confuse if not take a look on the code. I would suggest to correct "_the value return by nxsem_get_value()_" to "_the value 'sval' updated by nxsem_get_value()_"

**Additionally could somebody please explain why can_read() and can_write() enter_critical_section before accessing the received FIFO, but can_poll doesn't? Is it another bug and can_poll must be wrapped with critical section?**

Also probably it's better to recheck if nxsem_get_value used right in other places?
p.s. by the way there still one naming inconsistent 
https://github.com/apache/incubator-nuttx/blob/0e425584b8cd4e28705c6f0cf0da98be067f9ad3/arch/arm/src/stm32h7/stm32_spi_slave.c#L1553

lukegluke, 2020-07-03:
Other questions about can driver:
1. For what purpose is cd_nrxwaiters? In fact it used only for debug asserts
https://github.com/apache/incubator-nuttx/blob/57bc329aace6d816b9a51a368665ea4c7116a9c6/drivers/can/can.c#L1122
2. Why it is awkward in can_poll to be interrupted by a signal?
https://github.com/apache/incubator-nuttx/blob/57bc329aace6d816b9a51a368665ea4c7116a9c6/drivers/can/can.c#L1086-L1088

acassis, 2020-07-03:
Hi Olev,
Yes, you are right: https://pubs.opengroup.org/onlinepubs/9699919799/functions/sem_getvalue.html
Please submit a patch to fix the text.

The DEBUGASSERT to cd_nrxwaiters < 255 is there to guarantee we will have less than 255 threads reading the data.

Then can_poll receives a signal it will return the same why it returns when you get data, but then you need to check the errno var to verify if it returned with a value of EINTR. I think this is what the awkward means in this context.

lukegluke, 2020-07-03:
> Yes, you are right: https://pubs.opengroup.org/onlinepubs/9699919799/functions/sem_getvalue.html

If thoughts converge then they are limited :)

> The DEBUGASSERT to cd_nrxwaiters < 255 is there to guarantee we will have less than 255 threads reading the data.

I fully understand this. I just meant that if we are not debugging, cd_nrxwaiters modifications are done in vain.

> When the can_poll() receives a signal it will return in the same as it returns when you get data

In serial driver, uart_takesem can check returned value on EINTR, but it's still not used in uart_poll also with same awkward comment. But I'm not good in semaphores and poll processes, maybe it's all right and it works as should.

acassis, 2020-07-03:
Damn it, so we still limited, because you are right again!
This DEBUGASSERT is a case where it will not protect us in the wild. Should be replaced with ASSERT.

patacongo, 2020-07-03:
> The DEBUGASSERT to cd_nrxwaiters < 255 is there to guarantee we will have less than 255 threads reading the data.

There is no particular limit.  But cs_nrxwaiters is a uint8_t so this test is really only checking to assure that the uint8_t does not overflow.

ASSERTs in fielded code should be avoided.  Assertions are useful during debug but do not improve the quality of a fielded, buckled-up, embedded system.  They are product failures if they occur in a fielded product and only increase code size.

Embedded systems must be nearly error-free.  ASSERTs are errors.

lukegluke, 2020-07-06:
Encountered the new issue. For this time with tx semaphore. Please do not close this Issue.

lukegluke, 2020-07-06:
While investigating the issue in can, I've examined nxsem_wait and nxsem_post and didn't get how it suppose to work right for thread + thread case.

Semaphore can has two holders inside holder[2]. Let say two tasks have one thread each, and we are sync these threads by waiting on semaphore. CONFIG_SEM_PREALLOCHOLDERS is 0.

_nxsem_wait_ in first thread add "thread1" htcb to holder[0], then _nxsem_wait_ in second thread fill holder[1] with "thread2" htcb.
Then nxsem_post in first thread release holder[0] and 
> Check if there are any tasks in the waiting for semaphore task list that are waiting for this semaphore

 and found "task2" htcb (not the "thread2" htcb) and within _nxsem_addholder_tcb_ will fill holder[0]. Now holder[0] is occupied by htcb that will not get freed.
If situation repeats again, there will be no place for "thread2" -> DEBUGASSERT(pholder != NULL) in nxsem_allocholder.

Why if in nxsem_wait this_task() that return htcb of thread is used, in nxsem_post list of **tasks** waiting for semaphore is used, but not all threads?

I'm not good in shed staff, am I missing something?

lukegluke, 2020-07-06:
Regarding a bug with tx semaphore in CAN driver.
I write simple solution right away: **tx_sem must be SEM_PRIO_NONE**

Details.
`cd_xmit.tx_sem` is taken in `can_write()` and `can_poll()`, and is given in `can_txdone()`.
`can_txdone()` is called from MCU CAN interrupt from "Idle Task" (and in fact this is already enough to make semaphore SEM_PRIO_NONE)
Without this holders are used, but `nxsem_post()` in `can_txdone()` called from different thread can't free sem holders, so if you have two or more CAN writers and/or making CAN poll in different threads, it is matter of time to get into DEBUGASSERT(pholder != NULL) in nxsem_allocholder.

As for rx semaphore it also should be SEM_PRIO_NONE, because `nxsem_post()` in `can_receive()` is also called from MCU CAN interrupt. Only thing that's kept us from assertion on rx - rx_sem is unique for each reader (reader->fifo.rx_sem).

Speaking about rx_sem. We discuss DEBUGASSERT to cd_nrxwaiters < 255.
cs_nrxwaiters is a uint8_t, yes. But it is useless, we can just remove it.
In fact, no variable limits us with the number of readers, just an available memory. Readers are allocated on `can_open()` and are stored in a linked list.

There is only cd_ocount variable to count references to the port and it is uint8_t.
It is used only in `can_open()` and `can_close()`, but we could remove it and check cd_readers list instead (if it is empty). I would like to include this change in my patch, if nobody minds.

So if (for some only subjective reason) we want to limit the number of readers to 255, we can easily do it by checking list_length() of cd_readers list on `can_open()`. Please advise should I add this to the patch?

I don't know why I am so 'lucky' on semaphores assertions - 3 in a row in different places :)

closed 2020-07-14 by lukegluke