apache/nuttx #1368: graphics/vnc/server breakage

patacongo, 2020-07-04:
When the VNC server was created initially, it worked well.  However, in 2019 as part of the Twm4Nx development, I tried to use it and it did not work.  These are problems that I noted:

* Mouse Input is not received and, hence, the cursor is not drawn.  Twn4Nx is not useful with not mouse cursor.
* Background is black (it should be blue @16bpp).  Colors are bad.  Background is not transparent.  Toolbar buttons are visible but no other part of the Window.
* I think that root cause of the drawing errors is the RRE is not working.  Or, more likely, it has been deprecated and no longer supported by the host VNC client.
* There are other Twm4Nx/VNC server integration issues but these do not seem to be directly related to VNC server issues.

I did make several changes in graphics/vnc/server but did not fix the basic problems.

btashton, 2020-07-19:
@patacongo Is there a config that you were using for testing this?

patacongo, 2020-07-19:
> @patacongo Is there a config that you were using for testing this?

I tried to bring Twm4Nx (a Tom's Window Manager work-alike) using same70-xplained:twm4nx.

Previously, I was using samv71-xult:vnc and samv71-xult:vnxwm successfully in the past.  But I suspected those are broken as well too.

I did make several changes trying to bring up same70-xplained:twm4nx.  Since they were not successful they are suspect.

patacongo, 2020-07-19:
> I tried to bring Twm4Nx (a Tom's Window Manager work-alike) using same70-xplained:twm4nx.

My gut feeling was that there is some issue with the message called RRE which paints a rectangular region with a run length encoded data.  I think that the RRE message was deprecated.. I am not sure.  That is the behavior that I saw; like the RRE was being ignored.

GooTal, 2023-07-20:
Hi, i tried to use Twm4Nx recentyly but failed. I enabled some configs from menuconfig. But the `CONFIG TWM4NX_TITLE_FONTID` is missing from releases 12.1 branch. And i can\`t build the image successfully.

By the way, is graphics available using qemu? I\`ve seen some platforms using graphics on real hardware though. I\`m not quite familiar with graphics. Thanks a lot.