apache/nuttx #1407: 'configure.sh sim:nsh menuconfig' do not show menuconfig UI

liuguo09, 2020-07-16:
$ ./tools/configure.sh sim:nsh menuconfig
The cmds above do not show menuconfig UI since configure.sh calls sethost.sh, in which it redirects output to /dev/null if params not 'V=1'.  So the menuconfig UI would never show up. So  I wonder if this usage is normal, should we do not redirect to /dev/null? @Ouss4 @xiaoxiang781216 

tools/sethost.sh
_if grep -q "V=1" <<< "$*" ; then
  make olddefconfig $* || { echo "ERROR: failed to refresh"; exit 1; }
else
  make olddefconfig $* 1>/dev/null || { echo "ERROR: failed to refresh"; exit 1; }
fi_

xiaoxiang781216, 2020-07-16:
How about we change to?
```
make olddefconfig $ || { echo "ERROR: failed to refresh"; exit 1; }
```
It isn't only fix the issue mentioned here, but also help people identify the problem if error happen in configuration phase. Many newbie don't know V=1 option and then can't get any cue from output.

Ouss4, 2020-07-16:
The reason the redirection was added is because the output is too much.
I personally always considered the last option of sethost.sh (and similar script) to be a make option, not a make target.  So I'm not used to calling menuconfig that way.
If you guys want to remove that redirection, I for one have no objection.  At least there is still --no-print-directory to limit the output a little.

xiaoxiang781216, 2020-07-16:
> The reason the redirection was added is because the output is too much.

Many newbie hit the failure at the configuration step, but there isn't any context ouptut by default. The default behaviour make people which don't familiar with build system hard to find the root cause. Many people(include mentor) report it's hard to get a successful build, we need give more hint here to help the newbie.

> I personally always considered the last option of sethost.sh (and similar script) to be a make option, not a make target. So I'm not used to calling menuconfig that way.
> If you guys want to remove [...]

Ouss4, 2020-07-16:
Even though we merged the related PR, I'm actually a bit torn about this.
First, it's only the output that's redirected.  An error should still be printed if configure.sh or sethost.sh fail. V=1 not only removes the redirection but also enables make debugging.
The second thing is I still don't know if "configure.sh sim:nsh menuconfig" is well documentation use of configure.sh
As said above, the arguments after the board config are supposed to be make options not make targets.
Do you have any practical use of this?

xiaoxiang781216, 2020-07-17:
> Even though we merged the related PR, I'm actually a bit torn about this.
> First, it's only the output that's redirected. An error should still be printed if configure.sh or sethost.sh fail. V=1 not only removes the redirection but also enables make debugging.

Yes, the error still output but without any context information(one line error message). Since the build system inovke make recurisively into the subfolder, you can't even locate which Makefile hit the error. Here is an example @hartmannathan hit:
https://groups.google.com/forum/#!searchin/nuttx/configure.sh|sort:date/nuttx/R72PpM_8alQ/oWPaNvaZEwAJ
tools/configure.sh? I'm getting a make do_olddefconfig error:
```
$ tools/configure.sh -l tm4c1294-launchpad/nsh
  Copy files
  Select CONFIG_HOST_LINUX=y
  Refreshing...
make: *** [do_olddefconfig] Error 1
Has anyone else experienced this? I'm continuing to investigate...
```
Nobody can locate the location with so little information.

> The second thing is I still don't know if "configure.sh sim:nsh menuconfig" is well documentation use of configure.sh
> As said above, the arguments after the board config are supposed to be make options not make targets.
> Do you have any practical use of this?

To hidden the configuration step, we create builld.sh script like this:
```
configure.sh $*
make $*
```
User can invoke ```build.sh sim:nsh menuconfig``` to do configuration, it's work well but people report recently that the script hang if he/she run this command on a clear repo. After debuging, we found that it's because the output is redirected to /dev/null.

Ouss4, 2020-07-17:
This all makes sense, my only concern is that people are used to the less verbose output.  I guess we should let in for a while and see if we get any feedback.
I also provided #1415 to remove the redirection from configure.c.  Looking at that file it reminded me that under native Windows the more verbose output is the default and only option (no redirection).

closed 2020-07-17 by xiaoxiang781216