apache/nuttx #1409: Hard fault in STM32H7 protected mode

hexaglow, 2020-07-16:
Configuration: `stm32h747i-disco:nsh`. The system hard-faults on boot, and I haven't been able to track down the cause. It seems to occur after a supervisor call for a `boardctl` but I can't see any obvious reason why. I suspect stack corruption but can't tell where it is happening. My `.config` is below, that should be enough to reproduce, I have no extra applications or code in this build.

[My .config file is here](https://gist.github.com/hexaglow/cb085ab2b1fe55e2ca02035bbe98e57a)

btashton, 2020-07-18:
@hexaglow unfortunately I do no have this board to be able to reproduce this on hardware (none of the h7 series).  If you attach the elf with debug symbols (and an updated fault report) I might be able to get a little more information.  I suspect you have already worked through some of this but there is this resource here as well: https://cwiki.apache.org/confluence/display/NUTTX/Analyzing+Cortex-M+Hardfaults

@povauboin You had some some board support work for this would you be able to provide any insight sine this is a standard configuration.

btashton, 2020-07-27:
@hexaglow I now have a NUCLEO-H745 so I will try the protected build this week.  I did not have any H7 hardware before, hopefully we can figure out what is going on.