apache/nuttx #1414: port nuttx to NXP imx.rt600

mycan12345, 2020-07-17:
about port nuttx to NXP imx.rt600, cortex-m33,up to 300Mhz, 4.5MB on chip SRAM

Ouss4, 2020-07-17:
Hi, could you elaborate a little?
Are you asking if there is a port or are you willing to do the port?
As far as I can tell there is no ARMv8 chip in the repo, however there is support for the ARMv8 architecture.
If @xiaoxiang781216 passes from here he can give us more details about the state of the ARMv8 port.

xiaoxiang781216, 2020-07-17:
Yes, we use armv8m porting internally, which is work stable as armv7-m.

mycan12345, 2020-07-18:
Yes, i want port nuttx to NXP imx.rt600, but i dont konw how to start. @xiaoxiang781216 may i ask which platform do you guys are ported,thanks

xiaoxiang781216, 2020-07-18:
Realtek AMEBA ZII:
https://www.realtek.com/en/products/communications-network-ics/item/rtl8720cm

btashton, 2020-07-18:
@mycan12345 Are you planning on bringing it up with this EVK?
https://www.nxp.com/design/development-boards/i-mx-evaluation-and-development-boards/i-mx-rt600-evaluation-kit:MIMXRT685-EVK

I have the other EVKs for the imx.rt family, but I do not have this one.  I could be willing to track one down and help some if you are serious about the port.

patacongo, 2020-07-18:
@mycan12345 <https://github.com/mycan12345> Are you planning on bringing 
it up with this EVK?
>
> https://www.nxp.com/design/development-boards/i-mx-evaluation-and-development-boards/i-mx-rt600-evaluation-kit:MIMXRT685-EVK
>
> I have the other EVKs for the imx.rt family, but I do not have this 
> one. I could be willing to track one down and help some if you are 
> serious about the port.
>
Are the peripherals the same as for the i.MX RT?  If so, it would be 
good if the the i.MX 600 could cohabitate with the the i.MX RT in the 
same arch directory.

I used to forbid mixing CPU architectures in the same arch/arm/src/ MCU 
directories, but I have been doing that myself a few times lately.  We 
should try to follow a common patten.  You can see examples of this in:

    arch/arm/src/tiva, and
    arch/arm/src/s32k1xx

The s32k1xx is the more recent and the cleaner so that one would be the 
better model (although they both follow that same basic layout).  This 
directory supports the s32k11x (Cortex-M0+) and the s32k14x 
(Cortex-M4F).  Things to note:

  * Architecture specific source and header files reside in separate
    sub-directories, s32k11x/ and x32k14x/.
  * The Make.defs sets the VPATH and selects the C files based on the
    selected architecture:

    arch/arm/src/s32k1xx/Make.defs:

    110 # Source files specific to the ARM CPU family and to the S32K1xx
    chip family
    111
    112 ifeq ($(CONFIG_ARCH_CHIP_S32K11X),y)
    113 include s32k1xx/s32k11x/Make.defs
    114 else
    115 include s32k1xx/s32k14x/Make.defs
    116 endif

  * Common .c and .h files lie in the top-level arch/arm/src/s32k1xx/
    directory.  For the Tiva MCUs, the common source files list in
    arch/arch/src/tiva/common and the common .h files lie in the
    top-level arch/arch/src/tiva/ directory.  Tiva probably has that better.
  * The hardware/ sub-directory contains both common register definition
    files as well as MCU specific hardware definition files.

On the other hand, if the clock distribution, pin configuration, or 
critical peripherals are different then we should not consider that.  
Best to tolerate some duplication in that case and put the i.MX 600 in a 
separate directory altogether.

btashton, 2020-07-18:
Yeah. There is also the LPC55S69-EVK also from NXP which was the first one out with an M33 from them (I do have this board, just low on the priority list).  I noticed a fair bit shared hardware IP across the LPC and imx.rt lines, so even if we end up having to do this in its own arch folder, there is probably a lot of peripheral support to take inspiration from.

mycan12345, 2020-07-19:
@xiaoxiang781216 thank for share
@btashton yes, i got a MIMXRT685-EVK board,i want port nuttx to this board and run some POSIX compatible audio applications. if you can help me that very appreciate. 
@patacongo As I know the peripherals of rt600 is more like lpc54xx/lpc55xx

btashton, 2020-07-21:
@mycan12345 I should have one of those EVKs by the end of the week, so I can help with that.  Someone also asked for help adding support for the LPC55S69 for a low cost board that will be launched on crowdsupply, so looks like we should end up with a few target boards with this architecture.

hinxx, 2020-07-22:
I'll have OKdo E1 Development Board in couple of days to play around with; https://www.okdo.com/p/okdo-e1-development-board/.

btashton, 2020-07-24:
@hinxx that would be great to add support for that.  I have both the LPCXpresso55S69 and the MIMXRT685-EVK on hand now, so I'll see if I can put at least the very early support in place for for one of these boards.

btashton, 2020-07-24:
@patacongo unfortunately after digging in a bit the imxrt500/600 chip family really does not look like the other imxrt.  Instead it looks a lot like the lcp55 just with a custom DSP added on.  The template the used for the manual for imxrt600 is even the lpc style manual.

I'm not sure I want to try and join the lpc55 and the imxrt5/6xx families in the same folder, that sounds really confusing, even if a lot of the IP is the same.  I think I am going to add lpc55xx and imxrt6xx_5xx (like lpc17xx_40xx), does that sound reasonable from a naming convention?  I don't really like it, but I don't have a better idea at this point.

bhargavshah1988, 2021-05-04:
Can anyone provide me an update on RT600 Family port ?

Also, do we have any other Cortex M33 port ?

btashton, 2021-05-04:
there is a very initial port to the STM32L552 which I have recently been messing with which has the M33.  I also have a branch for the lpc55xx which is similar to the imx.rt600.

These are NXP chips I have boards for that I want to support, but have not been super motivated to pick back up the work on. If someone is interested I probably could start in on it again.

bhargavshah1988, 2021-05-05:
@btashton Thanks for quick update. I have Rt6xx board. I an help to port on it. Can you share your M33 branch ?

AlmAck, 2024-01-12:
Hi, I'm also interested to know the status of this port. I ordered a MIMXRT595-EVK dev kit and I would like to start experimenting a bit. @btashton where we can find your branch of the lpc55xx where I can look for?