apache/nuttx #1488: Use fair spinlocks in SMP

patacongo, 2020-07-30:
The spinlock implementation is very simple.  It is just a simple first-come, first-served spinlock.

This kind of spinlock and have adverse effects in SMP because under certain conditions other CPUs can always get the spinlock and one CPU can randomly be delayed for a significant amount of time.

We should consider replacing a fair spinlock implementation that enforces first-in, first out spinlocks.  There are several fair spinlock algorithms.  Some established and in common use (like ticket locks), some simple and some complex, some straight-forward and some more controversial.  A first step would be to decide on an appropriate fair spinlock algorithm.

patacongo, 2020-07-30:
> We should consider replacing a fair spinlock implementation that enforces first-in, first out spinlocks.

Most fair spinlock algorithms try to maintain FIFO spinlock access.  I would think for a real time system, priority based access would lead to better control over response latencies.

mu578, 2023-06-14:
Yes that was the point about my comment in the other thread, i.e context switching overheads entering critical section ; there is something like the Lamport algorithm if I remember well.

TaiJuWu, 2023-09-08:
Hello. I try to implement fair spinlock and put on [this.](https://github.com/TaiJuWu/nuttx/blob/1452773ead23ffa814800733db266b52cd066715/sched/semaphore/spinlock.c#L400)
I try to test by this [file.](https://github.com/TaiJuWu/nuttx-apps/blob/fair_spinlock_test/examples/hello/hello_main.c)
The result like this
```
Thread 6 is exiting lock 4 time.
Thread 7 is enter lock 4 time.
Thread 7 is exiting lock 4 time.
Thread 6 is enter lock 5 time.
Thread 6 is exiting lock 5 time.
Thread 7 is enter lock 5 time.
Thread 7 is exiting lock 5 time.
Thread 8 is enter lock 1 time.
Thread 8 is exiting lock 1 time.
Thread 9 is enter lock 1 time.
Thread 9 is exiting lock 1 time.
Thread 8 is enter lock 2 time.
Thread 8 is exiting lock 2 time.
Thread 9 is enter lock 2 time.
Thread 9 is exiting lock 2 time.
Thread 8 is enter lock 3 time.
Thread 8 is exiting lock 3 time.
Thread 9 is enter lock 3 time.
Thread 9 is exiting lock 3 time.
Thread 8 is enter lock 4 time.
Thread 8 is exiting lock 4 time.
```
But I only test on RR scheduler and I am not sure my implementation is correct or not.
So I hope I can get some suggestions from community.
If you have any suggestions, i would be very grateful. 

PS. I did't implement preemptive for this fair spinlock.

TaiJuWu, 2023-09-17:
> Yes that was the point about my comment in the other thread, i.e context switching overheads entering critical section ; there is something like the Lamport algorithm if I remember well.
> 
> @patacongo for some references; an interesting usage of ticket spinlocks within linux kernel ; https://github.com/drouyang/pmtlock

I have read this paper. 
It seem not work on priority base spinlock.
But work on same priority process.
But our requirement is priority base.

mu578, 2023-09-17:
You could overlay priority groups, you'd need just not to use the builtin list + having something timeout-able because you'd need a time-slicing mechanism but it is tricky to achieve with a SCHED_RR policy.

closed 2023-10-06 by xiaoxiang781216