apache/nuttx #1494: NXP imxrt1064 dose not support

degree900, 2020-07-31:
I tried to run the nuttx system on the NXP imxrt1064 development board and found that it cannot be used normally. Check the data sheet and found that the imxrt106x series supported by nuttx uses an external flash, while imxrt1060 is a built-in flash with different mapping addresses. I tried to modify the FLASH address in the link script and in the startup code, but found that I could not solve this problem myself. Do you have any plans to further support the imxrt1064 chip?

acassis, 2020-07-31:
hi @degree900 I used the imxrt1060-evk with NuttX recently (two weeks ago) using the external QSPI Flash and I can confirm it is working. The big issue was to program the QSPI flash memory and the jump configuration. Unfortunately the documentation in the board's user manual is incomplete. If you have a JLink PRO (I think JLink Base doesn't support flashing the external flash) it will be easier to flash. I spent much time trying to figure-out how to flash this board. The native drag-n-drop flashing feature wasn't working, it was able to flash small files, but returned timeout for big files. I tried to summarize it here to help you: https://acassis.wordpress.com/2020/07/31/just-a-resume-of-my-experience-with-imxrt1060-evk/

While I was searching for a solution for imxrt1060_evk I found this post: https://community.nxp.com/thread/518836

"You can actually run the LPC-Link2 firmware on the onboard debug probe of the RT1064-EVK board, in place of the default DAP-Link firmware. For more details see:"

That points to this post:
https://community.nxp.com/community/mcuxpresso/mcuxpresso-ide/blog/2019/01/23/overview-of-using-the-mimxrt1060-evk-with-mcuxpresso-ide

It should be nice if OpenOCD could have support to flash iMX RT10xx, it could make your life easier.

degree900, 2020-08-03:
Thank you so much for getting your reply so quickly! There may be some errors in my previous description. I will correct it now: I currently use NXP MCUXpressIDE to download the bin file of nuttx, and there is no problem in the download process. But my problem is that nuttx can indeed run on imxrt1060, but not on the development board of imxrt1064. I think the main reason is that imxrt1060 uses the 8MB external flash on the QSPI1 interface, the starting address is 0x60000000, and the imxrt1064 uses the 4MB built-in flash on the QSPI2 interface,with the starting address is 0x70000000. Note that in addition to the different addresses, the QSPI interface used is also different. Have you ever tried to actually run the code on imxrt1064 instead of imxrt1060? Looking forward to your reply again

acassis, 2020-08-03:
Hi @degree900 thank you for explaining more details about the issue. I don't have a imxrt1064 here to test, but did you try to change all "ORIGIN = 0x60000000" to "ORIGIN = 0x70000000" for those scripts at nuttx/boards/arm/imxrt/imxrt1060-evk/scripts/ ? Maybe it could work (or could not case there are more differences in these two boards). So, try to do it and let me know if you succeed.

degree900, 2020-08-03:
Yes, I tried to modify the link script, and also modified the FLASH_BASE and FLASH_END  at nuttx/configs/imxrt1060-evk/src/imxrt_flexspi_nor_boot.h  , but it still did not work

degree900, 2020-09-15:
I have solved this problem. In addition to modifying the flash address in the link script at nuttx/boards/arm/imxrt/imxrt1060-evk/scripts/ , I also need to change the LUT table in the file config/imxrt1060-evk/src/imxrt_flexspi_nor_flash.c

gauravsharma-bc, 2021-02-09:
Hi @degree900 , I'm on the same path as you- what did you change in the LUT in the 'imxrt_flexspi_nor_flash.c'? I don't really see anything address or peripheral specific there.

v01d, 2021-03-30:
I will close this since it appears that the issue reported by the OP was solved

closed 2021-03-30 by v01d