apache/nuttx #1511: Commit 60fe0a0f96613e7bc9ce5c6bb018f48638ca69f1 broke c++ builds.

davids5, 2020-08-04:
There was an unintentional side effect of 60fe0a0f96613e7bc9ce5c6bb018f48638ca69f1 

[image]

The very common variable name index can not be use with this change,
```
c ../../src/drivers/uavcan/libuavcan/libuavcan/src/node/uc_service_client.cpp
In file included from ../../src/drivers/uavcan/libuavcan/./libuavcan/include/uavcan/node/service_client.hpp:9,
                 from ../../src/drivers/uavcan/libuavcan/libuavcan/src/node/uc_service_client.cpp:5:
../../src/drivers/uavcan/libuavcan/./libuavcan/include/uavcan/util/multiset.hpp:145:33: error: macro "index" requires 2 arguments, but only 1 given
  145 |             : index(target_index)
```      |                                 ^

xiaoxiang781216, 2020-08-04:
Should we remove these two macros?

davids5, 2020-08-04:
Do you mean fix the api so that they are functions or just delete the macros?

closed 2020-08-05 by davids5