apache/nuttx #1539: SPIFFS and SmartFS must support . and .. entries

patacongo, 2020-08-08:
SPIFFS and SmartFS should support the standard Unix . and .. hardlinks (current directory and parent directory).  This is standard Unix behavior and should be supported by ALL NuttX file systems.

xiaoxiang781216, 2020-08-09:
@patacongo from my test show that not only spiffs and smartfs, but also binfs, fat, nxffs, procfs, and tmpfs. doesn't support '.' and '..'. And the implementation of cromfs is wrong: https://github.com/apache/incubator-nuttx/issues/1540

xiaoxiang781216, 2020-08-09:
BTW, it doesn't require to implement "." and ".." by adding the real entry into the file system, we can simulate them from readdir callback, like what Linux done:
https://github.com/torvalds/linux/search?q=%21dir_emit_dot&unscoped_q=%21dir_emit_dot

casaroli, 2026-11-08:
Possibly fixed by #19991.