apache/nuttx #1540: CROMFS implement "." and ".." in the wrong

xiaoxiang781216, 2020-08-09:
"." and ".." shouldn't set LINK flag because LINK flag is only used for the soft link, not the hard link.

xiaoxiang781216, 2020-08-09:
BTW, it doesn't require to implement "." and ".." by adding the real entry into the file system, we can simulate them from readdir callback, like what Linux done:
https://github.com/torvalds/linux/search?q=%21dir_emit_dot&unscoped_q=%21dir_emit_dot

patacongo, 2020-08-09:
Improving a correcting features is welcome.  Removing them is totally unacceptable without broad concurrence from the team.

patacongo, 2020-08-09:
There are lots of other issues with CROMFS hard links that also need to be addressed.  The logic follows the hard links seems to be missing.  It can probably be cloned from ROMFS function romfs_followhardlinks();

patacongo, 2020-08-10:
The correct fix is not simply to change the flags, but instead to implement following the hardlinks.  Both "." and ".." are hardlinks in CROMFS.  For all hardlinks, the file system should follow the hardlink which may refer to a directory, a file, or another hardlink.  When this finally resolves to a a directory or a file, then the correct type should be reported which may or may not a directory.  In the case of the hardlinks "." and ".." will, of course be a directory, but it may not be a directory in the general hardlink case.

I verified this logic in ROMFS (and fixed a bug in ROMFS too).  I will port the hard-link following logic into CROMFS in the next few days.  See PR #1546 ... I will make CROMFS work just like this.  I am only working part time these days, so it will take me some time to accomplish this.

Other file systems that don't support hardlinks can only fake the "." and ".." entries.

patacongo, 2020-08-10:
> "." and ".." shouldn't set LINK flag because LINK flag is only used for the soft link, not the hard link.

@xiaoxiang781216 The root pseudo-filesystem does support soft links and should return S_IFLNK.  Is this okay?

xiaoxiang781216, 2020-08-11:
> > "." and ".." shouldn't set LINK flag because LINK flag is only used for the soft link, not the hard link.
> 
> @xiaoxiang781216 The root pseudo-filesystem does support soft links and should return S_IFLNK. Is this okay?

The soft link(the file data contain the path to the target) return S_IFLNK, the hard link(inode directly point to the target block) should resolve the link internally(don't set S_IFLNK). So, it's right behaviour that the  root pseudo-filesystem return S_IFLNK because the link implemented by root pseduo filesystem contain the target path.

xiaoxiang781216, 2020-09-27:
Fixed by #1554

closed 2020-09-27 by xiaoxiang781216